MarlinFirmware/Marlin #22708: Latest bugfix (Sept.3, 2021) has blank screen, but not in Maple environment

rq3, 2021-09-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling latest bugfix with example configurations for the Trigorilla Pro goes fine, with no errors. Howver, when installed, the display is still dead.

HOWEVER, if compiled using the "Autobuild Marlin" extension in PlatformIO, and choosing the Trigorilla Pro Maple environment, the compiled firmware.bin works fine. Display is proper, and touchscreen works as expected.

Obviously, there is a fatal disconnect somewhere between the two build environments,

### Bug Timeline

Old. I never could get any Marlin after 2.0.9 to work if not built using the Maple environment.

### Expected behavior

I expect compiling the latest bugfix with the latest configuration files to yield a functional firmware.bin file under any available environment.

### Actual behavior

Only the Maple environment yields a functional compile.

### Steps to Reproduce

1) Compile and load Marlin under the standard Trigorilla Pro environment
2) Witness a dead display and touch screen
3) Compile and load Marlin under the Trigorilla Pro Maple environment
4) Witness a fully functional display and touchscreen

### Version of Marlin Firmware

latest bugfix (9/3/2021)

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla Pro

### Add-ons

none

thinkyhead, 2021-09-06:
Continue to use the Maple environment. The correct `trigorilla_pro` environment has simply not been established.

ellensp, 2021-09-06:
Does the non maple build environment actually boot? can you talk to it over serial?

rq3, 2021-09-09:
> 
> 
> Continue to use the Maple environment. The correct `trigorilla_pro` environment has simply not been established. Hopefully someone with the Trigorilla Pro board will figure it out and send us the corrections.

I have the board, schematics, and pinout. Specifically, what do you need?

rq3, 2021-09-09:
> 
> 
> Does the non maple build environment actually boot? can you talk to it over serial?

No. It does not boot or communicate over serial USB. The Maple environment boots and communicates over USB.

ellensp, 2021-09-09:
You need to work out whats different from the 
[env:trigorilla_pro_maple] to [env:trigorilla_pro]
Other than the change from 

platform          = ststm32@~12.1
board_build.core  = maple
to
platform         = ststm32@~12.1
board_build.core = stm32

normally you would attach a debugger to the controller and find out where it stopping, then work out why

ellensp, 2021-09-09:
Just had a little look, first thing I see is that env:trigorilla_pro_maple has the vector table at 0x08000000
but env:trigorilla_pro puts its at 0x08007000

In ini/stm32f1.ini is the following
```
[env:trigorilla_pro]
platform             = ${stm32_variant.platform}
extends              = stm32_variant
board                = genericSTM32F103ZE
board_build.variant  = MARLIN_F103Zx
board_build.offset   = 0x7000
build_flags          = ${stm32_variant.build_flags}
                       -DENABLE_HWSERIAL3 -DTIMER_SERIAL=TIM5
build_unflags        = ${stm32_variant.build_unflags}
                       -DUSBCON -DUSBD_USE_CDC

```
Try changing board_build.offset   = 0x7000 to board_build.offset   = 0x0000

This moves vector table back to 0x08000000

thinkyhead, 2021-09-10:
I'm not entirely clear on whether the `board_upload.offset_address` / `board_build.offset` options apply to both the `maple` and the `stm32` environments…. Need to look closer at that.

rq3, 2021-09-10:
> 
> 
> Just had a little look, first thing I see is that env:trigorilla_pro_maple has the firmware at 0x08000000
> but env:trigorilla_pro puts its at 0x08007000
> 
> In ini/stm32f1.ini is the following
> 
> ```
> [env:trigorilla_pro]
> platform             = ${stm32_variant.platform}
> extends              = stm32_variant
> board                = genericSTM32F103ZE
> board_build.variant  = MARLIN_F103Zx
> board_build.offset   = 0x7000
> build_flags          = ${stm32_variant.build_flags}
>                        -DENABLE_HWSERIAL3 -DTIMER_SERIAL=TIM5
> build_unflags        = ${stm32_variant.build_unflags}
>                        -DUSBCON -DUSBD_USE_CDC
> ```
> 
> Try changing board_build.offset = 0x7000 to board_build.offset = 0x0000
> 
> This moves the firmware back to 0x08000000

Downloaded the latest bugfix and associated configuration files for the Trigorilla Pro.
Changed the offset to 0x0000.
Does not compile:    error: the value of 'SystemCoreClock' is not usable in a constant expression
Also will not compile under the Maple environment.

ellensp, 2021-09-10:
That makes zero sense, as that variable only effects env:trigorilla_pro it doesn't touch env:trigorilla_pro_maple. Also I compiled it fine.

ellensp, 2021-09-10:
You keep saying " configuration files for the Trigorilla Pro." but marlin has configuration for Printers and board options for those printers. The Only Config files that uses this board that I can see are  CONFIG_EXAMPLES_DIR "delta/Anycubic/Predator"  Which is why I've been using for testing.

rq3, 2021-09-10:
> 
> 
> That makes zero sense, as that variable only effects env:trigorilla_pro it doesn't touch env:trigorilla_pro_maple. Also I compiled it fine.

I repeat. I downloaded the latest bugfix and the latest configuration files. It will not compile whether maple or not. I applied the offset of 0x0000 you suggested only to the non-maple ini. I agree that maple should not effect the standard environment, but now it does.

ellensp, 2021-09-10:
I always the the latest bugfix, freshly pulled every compile and cannot replicate this error.

ellensp, 2021-09-11:
delete your Marlin/.pio folder and try again. If it still fails attach the build log in its entirety.  (zip it up and drop zip into message, also include your Config files.)

rq3, 2021-09-11:
> 
> 
> You keep saying " configuration files for the Trigorilla Pro." but marlin has configuration for Printers and board options for those printers. The Only Config files that uses this board that I can see are CONFIG_EXAMPLES_DIR "delta/Anycubic/Predator" Which is what I've been using for testing.

That is also what I have been using for testing.

rq3, 2021-09-11:
> 
> 
> delete your Marlin/.pio folder and try again. If it still fails attach the build log in its entirety. (zip it up and drop zip into message, also include your Config files.)

@ellensp, attached is a zip file containing the build logs for the "standard" environment, and the "Maple" environment (which recently worked but now does not). I deleted the .pio directory and adjusted the offset in the standard environment to 0x0000 per your request.

As mentioned, the configuration files are the same ones that you are using, located in the CONFIG_EXAMPLES_DIR "delta/Anycubic/Predator" directory, with no modifications.
[Trigorilla_Pro_Buildlog.zip](https://github.com/MarlinFirmware/Marlin/files/7148062/Trigorilla_Pro_Buildlog.zip)

ellensp, 2021-09-12:
Did you not see all the warinings at the start

```
warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'buildroot\share\PlatformIO\scripts\common-dependencies.py'
File "C:\Users\rquin\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\common-cxxflags.py'
File "C:\Users\rquin\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\preflight-checks.py'
File "C:\Users\rquin\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\stm32_serialbuffer.py'
File "C:\Users\rquin\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175, in <module>

warning: Ignoring missing SConscript 'buildroot\share\PlatformIO\scripts\generic_create_variant.py'
File "C:\Users\rquin\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175, in <module>
```

ellensp, 2021-09-12:
check that these files exsist 
buildroot\share\PlatformIO\scripts\common-dependencies.py
buildroot\share\PlatformIO\scripts\common-cxxflags.py
buildroot\share\PlatformIO\scripts\preflight-checks.py
buildroot\share\PlatformIO\scripts\stm32_serialbuffer.py
buildroot\share\PlatformIO\scripts\generic_create_variant.py
etc

rq3, 2021-09-12:
> Did you not see all the warinings at the start
>
> ```
> warning: Calling missing SConscript without error is deprecated.
> Transition by adding must_exist=0 to SConscript calls.
> Missing SConscript [...]

@ellensp, of course I saw them. Are you saying now that you now can't comile also? What am I supposed to do with these warnings?
You and I are both compiling from the same bugfix, with the same configuration files, for the same printer. The compile fails, with or without your offset to 0x0000 correction.

Tell me PRECISELY what you need.

ellensp, 2021-09-12:
I do not get those warning, only you are having issues compiling. 
I need to to check that you have the files listed above 
I suspect you have moved or deleted them.

rq3, 2021-09-12:
> 
> 
> I do not get those warning, only you are having issues compiling.
> I need you to check that you have the files listed above.
> I suspect you have moved or deleted them.

@ellensp, my apologies. I deleted the entire existing Marlin file, and downloaded fresh copies of both the latest bugfix and 2.0.9.1, along with their associated configuration files for the Anycubic Predator. After making sure that the offset in STM32F1.ini was set to 0x0000, and that the directory structure was intact, these are the results:
1) bugfix compiles with no problems or errors and functions in the printer for both standard and maple environments.
2) For what it's worth, 2.0.9.1 does not compile in the standard environment, but does compile and functions in the maple environment.

ellensp, 2021-09-12:
No. initial bug is real, just lost of superfluous issues when testing the fix. Creating PR with offset correction

rq3, 2021-09-12:
> 
> 
> No. initial bug is real, just lost of superfluous issues when testing the fix. Creating PR with offset correction

Thank you for your efforts and diplomacy! It is interesting to note that, for the first time, this printer and its display can now use the Color_UI instead of the Classic_UI. I had to disable the mesh overlay to get it to compile, but it looks very nice on the printer.

closed 2021-09-12 by rq3