MarlinFirmware/Marlin #22711: [BUG] Not possible currently to compile marlin 2.0.9.1 using only the arduino IDE

Wallby, 2021-09-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Since U8glib-HAL 0.5.0 U8glib.h was renamed to U8glib-HAL.h (https://github.com/MarlinFirmware/Marlin/commit/69b44c2309d859865d4724cb8e323a13ba535d3c).

As there is only one version (0.5.0 it seems) of U8glib-HAL available via the arduino library manager..
[image]
.. it is not possible atm the get marlin 2.0.9.1 to compile using only arduino IDE (marlin 2.0.9.1 seems to require U8glib-HAL 0.4.5 or earlier).

NOTE: I did not test the bugfix-2.0.x code, but this issue is not about the bugfix-2.0.x branch
NOTE: I have not tested if 0.4.5 actually works, but as https://github.com/MarlinFirmware/Marlin/commit/c9a7fd1722764f2b790a38f2c7ee0c4d00d77c17 did make it into 2.0.9.1, but https://github.com/MarlinFirmware/Marlin/commit/69b44c2309d859865d4724cb8e323a13ba535d3c did not.. it seems like 2.0.9.1 was intended to be used with U8glib-HAL 0.4.5

### Bug Timeline

See description (it seems that this bug is caused by not providing the option to download 0.4.5 via the arduino library manager)

### Expected behavior

I am able to select 0.4.5/0.5.0 via the arduino library manager + there is a note in the readme.md of marlin "explaining the situation".

### Actual behavior

See bug description

### Steps to Reproduce

1. Extract marlin 2.0.9.1 in a folder
2. Open Marlin.ino in Arduino IDE
3. Install U8glib via the library manager (sketch -> "use library" -> "manage libraries") //< not sure what these options are actually called as display language is not english for me
4. Press the "☑" to compile

### Version of Marlin Firmware

2.0.9.1

### Additional information & file uploads

Other people experiencing this issue..
https://github.com/MarlinFirmware/Marlin/issues/22652
https://github.com/MarlinFirmware/Marlin/pull/22324#issuecomment-885567806

Wallby, 2021-09-04:
> [image]
> [image]

NOTE: even though in these images U8glib seems to be installed and U8glib-HAL not, these images are merely for showing that there is no option to select a version of U8glib-HAL.

tpruvot, 2021-09-04:
unsure its the reason but not all arch are set in the library properties...
 https://github.com/MarlinFirmware/U8glib-HAL/blob/master/library.properties#L9

ellensp, 2021-09-05:
There is only one version of U8glib-HAL version 0.5.0

ellensp, 2021-09-05:
You do not need U8glib at all, you only need  U8glib-HAL for newer marlin

ellensp, 2021-09-05:
The features.ini in 2.0.9.1 specifies U8glib-HAL@~0.4.5  for platformio. It is also required by Arduino ide.

This is a modified U8glib from Marlin.  It is in the MarlinFirmware/U8glib-HAL repository, but internally it is still called U8glib.h (this is why you cannot install it from the arduino ide library manager, it conflicts with the current U8glib) 
You get it from here. https://github.com/MarlinFirmware/U8glib-HAL/releases/tag/0.4.5

In bugfix and in next release Marlin it is fully renamed to U8glib-HALh, to try and make it obvious that you should not use U8glib from arduin ide library manager.

To use the correct version do the following.
Uninstall standard U8glib from the Arduino IDE library manager.
Download U8glib-HAL from https://github.com/MarlinFirmware/U8glib-HAL/releases/tag/0.4.5
Extract the directory into your Arduino/libraries folder
Download https://github.com/MarlinFirmware/Marlin/files/6710738/library.properties.zip
Extract library.properties to Arduino/libraries/U8glib-HAL-0.4.5
Restart Arduino IDE so it notices libraries have changed.
Click build or upload  and take a long nap while arduino ide slowly compiles.

Or are has been mentioned a billion times, give up on the old criminally backwards and slow Arduino IDE and use platfomio/vscode. 
It install all required library automatically and is about 10 time faster to compile Marlin.

closed 2021-09-05 by ellensp

ellensp, 2021-09-05:
"Did you test the latest bugfix-2.0.x code? Yes, and the problem still exists."
This is incorrect

You simply install U8glib-HAL v 0.5.0 from the arduino ide library manager and compile.

Wallby, 2021-09-05:
@ellensp 

> This is incorrect

No it is not. The problem is that "It is not possible currently to compile marlin 2.0.9.1 using only the arduino IDE" (i.e. 2.0.9.1 doesn't work with 0.5.0).

ellensp, 2021-09-05:
The problem does not exist in bugfix.

Wallby, 2021-09-05:
What's with the emoji? It might be that bugfix compiles, but compiling marlin 2.0.9.1 (an official release) using only the arduino IDE is not possible, hence "It is not possible currently to compile marlin 2.0.9.1 using only the arduino IDE" is not false.