MarlinFirmware/Marlin #22712: [BUG] Fade height causing issues with bed adhesion.

quickfeet14, 2021-09-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While the fade height option is enabled/uncommented, both bilinear and UBL don't compensate enough on parts of the bed. Mesh validation (G26) lays down a beautiful print showing the bed adhesion, however when I go to print a part, for half of the bed the nozzle is way above the build surface and won't adhere to the bed at all.

I have used both the RESTORE_BED_LEVELING_AFTER_G28 and ENABLE_BED_LEVELING_AFTER_G28 to make sure the bed leveling is on, as well as adding a G29 A/M420 S1 to my start gcode after the last G28. The only way to get the nozzle to obey the correct z-offset is to disable the fade height option in the firmware.

I currently have the setting off in the firmware and it prints well enough, so it is not a time sensitive issue. Mostly posting this for other people with the same problem.

### Bug Timeline

Began as I upgraded from 1.x to 2.0.9.1

### Expected behavior

Bed adhesion while fade height is enabled

### Actual behavior

No bed adhesion while fade height option is enabled

### Steps to Reproduce

Enable fade height in firmware.
Print does not stick to bed.
Disable fade height in firmware.
Print sticks.

### Version of Marlin Firmware

2.0.9.1

### Printer model

D-Bot CoreXY

### Electronics

Ramps 1.4 on a Mega

### Add-ons

BLtouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/7110446/Configs.zip)

ais80, 2021-09-07:
Same issue. With BLTouch on Bilinear and UBL. Also with manual mesh levelling, but virtually no bed adhesion with BLTouch and ABL enabled. Uncommenting enabling fade height causes bed adhesion to be restored and my first print in 2 weeks

thinkyhead, 2022-04-21:
If bed adhesion is an issue, try increasing the fade distance to 10 or higher. For an 0.2mm first layer that will result in only 1/50th of the bed leveling compensation being applied, and then 1/25 on the next layer, and so on, which should be pretty negligible when the amount of compensation is relatively small. Definitely make sure to tram the bed, and definitely make sure to heat the bed and let it sit before doing `G29`.