MarlinFirmware/Marlin #22714: [BUG] (bug summary)

r4ph488, 2021-09-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi Guys,
I use an idex .
Everything Wirkung fine.

I have a Problem with e1.
e0 heats Up ans can Extrude.

E1 heats Up ans sonst Extrude.

I have Changes Extruder / nozzle from 1 to 2
I have defined e1 tmc2209

When i will Extrude the Motor dont work.i have tested an Others Motor still the Same Problem in thus Port.
M122 says Connection e1 ....OK
I tested changing e1 to E3 Port Problem ist the Same.
Pinout ist ok.

Has someone an Idea where ist the Problem ?

I use the bigtreetech octopus Board.

### Version of Marlin Firmware

Marlin 2.0.9.1

ellensp, 2021-09-05:
Comment out  #define DISABLE_INACTIVE_E In Configuration_adv.h

r4ph488, 2021-09-05:
Thank you i will Check and give Feedback.

r4ph488, 2021-09-05:
Okay ellensp ... that works fine :)
can you help my by my last problem too ?

i have one stepper port free ... but when i disable exp mot vsc says no e plug left for z3 but i have one free ... do you now where i can cahnge it ? or i can set z3 for ... e?..... ?

thank you dude

ellensp, 2021-09-05:
octopus,  It can have 8 stepper drivers
It comes configured as X, Y, Z, Z2, E0, E1, E2, E3

Do you mean BTT_MOTOR_EXPANSION?  With gives you E4, E5, E6 ?
Unused extruders can be reallocate for other axies.

Without knowing what your #define EXTRUDERS is set to and what you have plugged into where... I cant really say.

r4ph488, 2021-09-05:
I think i dont need exp mot... Should i Upload my configuration.h and ADV ?

r4ph488, 2021-09-07:
[image]
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7122766/Marlin.zip)

[image]

So i have uploaded my cfg files and a picture from my board and motor connection.  
i want set z3 @ picture but when i turn off btt exp mot vsc tell me there is no e plug for z3.
where can i set my z3 to e3 ?

`In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:811,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins_postprocess.h:773:8: error: #error "No E stepper plug left for Z3!"
  773 |       #error "No E stepper plug left for Z3!"
      |        ^~~~~`

thank you :)

ellensp, 2021-09-14:
There is an issue with the code.
Since the octopus already has a Z2 pins defined, the reuse allocated E as Zx code is getting confused. 
Its still allocating a E to be for Z2 when it shouldnt..
Working on a fix

ellensp, 2021-09-14:
Can you try this diff ?  It compiles, and it should work, but needs testing
```DIFF
diff --git a/Marlin/src/pins/pins_postprocess.h b/Marlin/src/pins/pins_postprocess.h
index 6b5c695e85..94336429b7 100644
--- a/Marlin/src/pins/pins_postprocess.h
+++ b/Marlin/src/pins/pins_postprocess.h
@@ -697,14 +697,6 @@
 
 // The Z2 axis, if any, should be the next open extruder port
 #if NUM_Z_STEPPER_DRIVERS >= 2
-  #ifndef Z2_STEP_PIN
-    #define Z2_STEP_PIN   _EPIN(Z2_E_INDEX, STEP)
-    #define Z2_DIR_PIN    _EPIN(Z2_E_INDEX, DIR)
-    #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE)
-    #if Z2_E_INDEX >= MAX_E_STEPPERS || !PIN_EXISTS(Z2_STEP)
-      #error "No E stepper plug left for Z2!"
-    #endif
-  #endif
   #ifndef Z2_MS1_PIN
     #define Z2_MS1_PIN    _EPIN(Z2_E_INDEX, MS1)
   #endif
@@ -746,9 +738,17 @@
     #endif
     #undef Z2_DIAG_PIN
   #endif
-  #define Z3_E_INDEX INCREMENT(Z2_E_INDEX)
-#else
-  #define Z3_E_INDEX Z2_E_INDEX
+  #ifndef Z2_STEP_PIN
+    #define Z2_STEP_PIN   _EPIN(Z2_E_INDEX, STEP)
+    #define Z2_DIR_PIN    _EPIN(Z2_E_INDEX, DIR)
+    #define Z2_ENABLE_PIN _EPIN(Z2_E_INDEX, ENABLE)
+    #if Z2_E_INDEX >= MAX_E_STEPPERS || !PIN_EXISTS(Z2_STEP)
+      #error "No E stepper plug left for Z2!"
+    #endif
+    #define Z3_E_INDEX INCREMENT(Z2_E_INDEX)
+  #else
+    #define Z3_E_INDEX Z2_E_INDEX
+  #endif
 #endif
 
 #ifndef Z2_CS_PIN
@@ -773,6 +773,7 @@
       #error "No E stepper plug left for Z3!"
     #endif
   #endif
+
   #if AXIS_HAS_SPI(Z3)
     #ifndef Z3_CS_PIN
       #define Z3_CS_PIN   _EPIN(Z3_E_INDEX, CS)

```

ellensp, 2021-09-14:
Can you test this PR, It should work.

r4ph488, 2021-09-14:
does the bugfix now fix the problem?

ellensp, 2021-09-16:
A new for the "No E stepper plug left for Z3!" issue has been added to bugfix. Give the current bugfix a try.

r4ph488, 2021-09-16:
Yes now all Stepper in the octopus i dont need the Extension Board more...
But now e1 has Power but dont step.

ellensp, 2021-09-16:
enable #define PINS_DEBUGGING in Configuration_adv.h 
re Compile and upload
Send the Machine a M43 and capture the output. 
Zip up the output and attach in a reply
Looking for any weirdness in E1 pins...

r4ph488, 2021-09-16:
[debugging.txt](https://github.com/MarlinFirmware/Marlin/files/7179231/debugging.txt)

Ok i have attached ...

ellensp, 2021-09-17:
himm getting weird repeated values in that output.
LESEN: PIN: PC13        M42 P45          E1_STEP_PIN                            protected 
LESEN: PIN: PF0         M42 P80          E1_DIR_PIN                             protected 
LESEN: PIN: PF1         M42 P81          E1_ENABLE_PIN                          protected 
but it shows the correct pins according to pins diagram https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/blob/master/Hardware/BIGTREETECH%20Octopus%20-%20PIN.pdf

r4ph488, 2021-09-17:
OK and what can i do ?

r4ph488, 2021-09-18:
Ok i have tested ....
i have activated btt extension board
i have changed e1 ( motor 5 ) from my octopus board to e4 from btt extensionboard ... now e1 works ....
but not on the octopus board.... only motor port 5 dont works correct....
any idea where the problem is ?

thinkyhead, 2026-10-08:
This looks to have been addressed by #22771, merged since this report, and there's been no further activity here. Closing for now as part of 2.1.3 cleanup. If you still see the problem on current `bugfix-2.1.x`, please let us know and we'll reopen.

closed 2026-10-08 by thinkyhead