MarlinFirmware/Marlin #22716: [BUG] can't disable SDSUPPORT with DWIN_CREALITY_LCD

electronicm, 2021-09-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I disable SDSUPPORT and want to use DWIN_CREALITY_LCD, I get the following error message when compiling:
```
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\marlinui.cpp.o
Marlin\src\lcd\e3v2\creality\dwin.cpp: In function 'uint16_t nr_sd_menu_items()':
Marlin\src\lcd\e3v2\creality\dwin.cpp:406:10: error: 'card' was not declared in this scope
  406 |   return card.get_num_Files() + !card.flag.workDirIsRoot;
      |          ^~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\BL24CXX.cpp.o
Marlin\src\lcd\e3v2\creality\dwin.cpp:1251:23: error: 'card' was not declared in this scope
 1251 |   char * const name = card.longest_filename();
      |                       ^~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\buzzer.cpp.o
Marlin\src\lcd\e3v2\creality\dwin.cpp: In function 'void make_name_without_ext(char*, char*, size_t)':
Marlin\src\lcd\e3v2\creality\dwin.cpp:1784:23: error: 'card' was not declared in this scope
 1784 |   char * const name = card.longest_filename();
      |                       ^~~~
Marlin\src\lcd\e3v2\creality\dwin.cpp: In function 'void HMI_SDCardInit()':
Marlin\src\lcd\e3v2\creality\dwin.cpp:1806:25: error: 'card' was not declared in this scope
 1806 | void HMI_SDCardInit() { card.cdroot(); }
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\hex_print.cpp.o
      |                         ^~~~
Marlin\src\lcd\e3v2\creality\dwin.cpp: In function 'void Draw_SDItem(uint16_t, int16_t)':
Marlin\src\lcd\e3v2\creality\dwin.cpp:1843:27: error: 'card' was not declared in this scope
 1843 |   const bool is_subdir = !card.flag.workDirIsRoot;
      |                           ^~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\creality\dwin.cpp.o] Error 1
```

### Steps to Reproduce

Compile FW for `#define MOTHERBOARD BOARD_CREALITY_V4` with `//#define SDSUPPORT` and `#define DWIN_CREALITY_LCD`.

### Version of Marlin Firmware

bugfix-2.0.x (2021-09-05)

### Printer model

Creality Ender 3v2

### Bed Leveling

UBL Bilinear mesh

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7111841/Configuration.zip)

VSCode with `Auto Build Marlin v2.1.34` & `PlatformIO IDE v2.3.3`

ellensp, 2021-09-05:
Since that LCD runs its own firmware, it cannot be reconfigured on the fly.  Thus it has requirements. One is you NEED to have support enabled.

electronicm, 2021-09-05:
@mriscoc: is that someting for you?
#21942

electronicm, 2021-09-05:
> Since that LCD runs its own firmware, it cannot be reconfigured on the fly. Thus it has requirements. One is you NEED to have support enabled.

I've been testing @The-EG's work #22211 for months and it's never been a problem. Only when I switched back to CrealityUI did it become a problem.

When I use DWIN_MARLINUI_* everything works.

The-EG, 2021-09-05:
> When I use DWIN_MARLINUI_* everything works.

The MarlinUI and 'Creality'UI implementations are completely separate. It would appear that the CrealityUI assumes that SDSupport always exists, where the MarlinUI implementation does not.

mriscoc, 2021-09-05:
> @mriscoc: is that someting for you?
> #21942

The CrealityUI assumes that you have a 3D printer from them, but could be easy for someone to put some #ifdef to disable the menus and functionalities not available without a SD card. The PR #22211 uses the native MarlinUI implementation used in normal LCD display, which is more generic, and can work without a SD card.

ellensp, 2021-09-05:
You can disable the need for sdsupport in marlin, but all the SDsupport related icons and features will still be on screen. Resulting in bug reports that the icons/features don't work!

thisiskeithb, 2021-09-06:
There's now a sanity check to alert users of the required settings: https://github.com/MarlinFirmware/Marlin/commit/b0a91073b25f53e9e377f0ef4e01c9dd6fefd021

> #error "DWIN_CREALITY_LCD requires SDSUPPORT to be enabled."

closed 2021-09-06 by thisiskeithb

thinkyhead, 2021-09-07:
The better options for maximum flexibility are `DWIN_MARLINUI_PORTRAIT` or `DWIN_MARLINUI_LANDSCAPE`.

electronicm, 2021-09-08:
> The better options for maximum flexibility are `DWIN_MARLINUI_PORTRAIT` or `DWIN_MARLINUI_LANDSCAPE`.

As for flexibility, I totally agree with you.
However, the look of the CrealityUI appeals to me more and it has other advantages for me:
I like the CrealityUI optically better and it has some other advantages:
- In order to move the axes, the speed does not have to be selected in advance.
- Permanent visibility of some system information (position, temperature, status text)

Could they be transferred to the MarlinUI?