MarlinFirmware/Marlin #22717: [BUG] layer shift on z axis

exverse, 2021-09-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After updating to marlin 2, first using bugfix branch started getting random layer shifts on z axis. I have checked tmc drivers vref, even swapped them and still only on y axis. Belts are good. The layer shift is random using same gcode. Seems like there was an issue in marlin causing this but it was closed so i figured it was solved.

### Bug Timeline

Started soon as i updated marlin

### Expected behavior

Not to shift layers

### Actual behavior

Shifting layers on z axis

### Version of Marlin Firmware

2.0 bugfix branch

### Printer model

Toauto puramid a1.1

### Your Slicer

Simplify3D

thisiskeithb, 2021-09-06:
Try increasing the driver current, slowing down movement/printing/probing speeds, reducing jerk settings, etc.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-06 by thisiskeithb

exverse, 2021-09-06:
This seems to be a firmware issue, went to 2.0.8 with all the same settings with no issue.

lclrcarvalho, 2021-09-23:
I would like to say that I'm having exactly the same issue consistently with bugfix-2.0.x updated around a week ago.
I have attached pictures.
On the 1st print it wasn't evident that was some software issue as there was a large travel movement, so I dropped speeds but had no effect, and the layer shift happened basically on the same place.  
Tried a different print and on the 2nd case is very evident that the layer shift happens on the Y axe and while there are holes in X axes. When it the hole is completed the layer shift stops and continues the print in Y axe correctly.
On hindsight, on the first print, there is a layer where the Y axe didn't aligned properly, but is like one off...

As per previous comment, I'll try 2.0.8 to see if my problem also disappears.  Theoretically this could be due to the cura slicer as I recently upgraded to cura 4.11.0 just before these prints ...

A did a small test with a small piece, but had no issue.  The prints that have failed were printing fine for over 7~8 hours before the shift happened.

Half of a spool to the bin...

[image]
[image]