MarlinFirmware/Marlin #22720: [BUG] MKS MINI12864 Not working with SKR1.4/t

KetamineKyle, 2021-09-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I defined my display of choice, the mks mini 12864, Partnered with an SKR1.4T with TMC2130's drivers running in SPI mode, I've tried to reverse the connectors, All that I get is the screen glowing white as it getting fed the 5v it needs, It seems to be buzzing every time I click the Pot, but shows no characters, I verified that the display is functional by hooking it up to a TriGorilla 1.3 board and the Mks Gen L v1.0.

## Configuration.h
```cpp
#define CONFIGURATION_H_VERSION 02000901   
#define MKS_MINI_12864
```

## pins_BTT_SKR_V1_4.h
```cpp
#if ENABLED(MKS_MINI_12864)
  #define DOGLCD_CS            EXP1_05_PIN
  #define DOGLCD_A0            EXP1_04_PIN
  #define DOGLCD_SCK           EXP2_09_PIN
  #define DOGLCD_MOSI          EXP2_05_PIN
  #define FORCE_SOFT_SPI
#endif
```

### Bug Timeline

Today, as I tried to attach the screen to reuse it

### Expected behavior

The display is expected to, you know, display stuff.

### Actual behavior

2-3 Second buzzing noise when plugged in, the backlight turns on, audible noise when the pot is turned/clicked, seems like its working, just not displaying.it's

### Steps to Reproduce

1. TMC2130 SPI Paired with an SKR 1.4
2. Compile marlin with the `#define MKS_MINI_12864` display defined.
3. Flash.

### Version of Marlin Firmware

02000901   

### Printer model

Tevo Tarantula Pro

### Electronics

TMC 2130's, Dual Z, 24v, SKR 1.4 Turbo, stock display

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

ellensp, 2021-09-06:
"tried to reverse the connectors" that's a shame. That kills one of the io lines on the SKR1.4/t nearly every single time.

KetamineKyle, 2021-09-06:
> "tried to reverse the connectors" that's a shame. That kills one of the io lines on the SKR1.4/t nearly every single time.

Any quick way of testing that?

ellensp, 2021-09-06:
No easy way I know of.

Star from the beginning.
Check the following.  

Skr 1.4 (note cable key is on the left)
```
/**              ------                             ------
 *  BEEPER 1.30 |10  9 | 0.28 BTN_ENC         0.17 |10  9 | 0.15 SCK
 *         1.18 | 8  7 | 1.19         BTN_EN2 3.26 | 8  7 | 0.16 SDSS
 *         1.20   6  5 | 1.21 CS      BTN_EN1 3.25   6  5 | 0.18 MOSI
 *      A0 1.22 | 4  3 | 1.23          SD_DET 1.31 | 4  3 | RESET
 *          GND | 2  1 | 5V                    GND | 2  1 | NC
 *               ------                            ------
 *                EXP1                              EXP2
*/
```
MKS mini (note cable keys are on right)
[image]

So according to this you need to rotate the plugs around 180 degrees at one end..  while keeping exp1 going to exp1 and exp2 going to exp2. 

While off and unplugged from usb and power. Plug in the lcd the way you think it should be
With a multi meter  check continuity from controller gnd to lcd gnd (normally the metal sd card holder on the board to the lcd sd card is good for this)
Then check 5v on controller is connected to 5v on lcd.    
If those are fine, it should be safe to power up and try again

KetamineKyle, 2021-09-06:
I Followed that Pinout and assumed it was safe to flip both 180 as then the The GND and VCC align on EXP1, and GND aligns on EXP2

KetamineKyle, 2021-09-06:
Yep, I verified that It GND => GND and 5V => 5V, But still no text, The display seems like it's trying to refresh them as there's a bit of scrolling text at the bottom in mariln

thisiskeithb, 2021-09-06:
Which LCD revision do you have? If you could get a picture of the back, that’d help.

I’ll see which revs I have and will toss them on an SKR 1.4.

KetamineKyle, 2021-09-06:
Sure,

[image]
[image]

ellensp, 2021-09-06:
This page https://www.aliexpress.com/item/32870541074.html says it should be set to a MINIPANEL not a MKS_MINI_12864

thisiskeithb, 2021-09-06:
> This page https://www.aliexpress.com/item/32870541074.html says it should be set to a MINIPANEL not a MKS_MINI_12864

It’s also outdated.

KetamineKyle, 2021-09-06:
Yeah, I couldn't find much on the V2.0, Marlin has V1.0 and V3.0(bugfix branch) but no V2.0

KetamineKyle, 2021-09-06:
Also, If I try compile for the MINIPANEL, I get a nice message from marlin, In the bugfix branch and the release
```Marlin\src\lcd\dogm\marlinui_DOGM.h:225:23: error: 'DOGLCD_CS' was not declared in this scope; did you mean 'DOGLCD_SCK'?
  225 |     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
      |                       ^~~~~~~~~
Marlin\src\lcd\dogm\marlinui_DOGM.cpp:262:14: note: in expansion of macro 'U8G_PARAM'
  262 |     u8g.init(U8G_PARAM);
      |              ^~~~~~~~~
Compiling .pio\build\LPC1769\src\src\lcd\extui\ui_api.cpp.o
Marlin\src\lcd\dogm\marlinui_DOGM.h:225:34: error: 'DOGLCD_A0' was not declared in this scope; did you mean 'DOGLCD_SCK'?
  225 |     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
      |                                  ^~~~~~~~~
Marlin\src\lcd\dogm\marlinui_DOGM.cpp:262:14: note: in expansion of macro 'U8G_PARAM'
  262 |     u8g.init(U8G_PARAM);```

ellensp, 2021-09-06:
No pin definitions for MINIPANEL for skr 1.4 

change #if ENABLED(MKS_MINI_12864)  to  #if EITHER(FYSETC_MINI_12864, MINIPANEL)
In Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_4.h  It now compiles

KetamineKyle, 2021-09-06:
Flashed it, still just a white screen, the Nob works, and so dose the reset button, Just nothing gets displayed

ellensp, 2021-09-06:
try changing   
```CPP
#if ENABLED(FYSETC_MINI_12864, LPC1769)
      #define DOGLCD_CS              EXP1_08_PIN
      #define DOGLCD_A0              EXP1_07_PIN
```
```CPP
#if ENABLED(FYSETC_MINI_12864, LPC1769)
    #define DOGLCD_CS              EXP1_05_PIN
    #define DOGLCD_A0              EXP1_04_PIN

```

KetamineKyle, 2021-09-06:
nope, Do i just give up and buy a new display

thisiskeithb, 2021-09-06:
`MKS_MINI_12864` does other things like set the contrast and defines `MINIPANEL`, so just defining `MINIPANEL` won’t work.

KetamineKyle, 2021-09-06:
So what do you suggest? Enable `MKS_MINI_12864` and reconfigure the pins from `minipanel` to `MKS_MINI_12864`?

KetamineKyle, 2021-09-06:
Its so so strange, I ran G28 than G29 And I have a manual probe and the display beeps and all

thisiskeithb, 2021-09-06:
I’ve managed to temporarily lose my MKS Mini V2 screens in my pile-o-hardware, but I’m sure we can get this LCD working since it works fine on the SKR 1.3.

thisiskeithb, 2021-09-06:
So I have an MKS Mini 12864 V1.0 and V2.1.

Both of these work on an SKR 1.4 Turbo but the 1.0 seems to need different contrast settings.

Can you try setting your LCD contrast using g-code? `M250 C155` or `M250 C195` will change the [LCD contrast](https://marlinfw.org/docs/gcode/M250.html).

KetamineKyle, 2021-09-07:
I'll grab my printer and check

KetamineKyle, 2021-09-07:
I can change

KetamineKyle, 2021-09-07:
Clicked the wrong button, I can now change the contrast as beforehand `M250 CXXX`  Would not change the contrast, It would just stay the same, Are you using the stock Pin out of the board, I have a feeling this is a ` SKR1.4` Thing as my friends 1.3 works fine with his `mks mini12864 v2`

thisiskeithb, 2021-09-07:
> I can now change the contrast as beforehand M250 CXXX Would not change the contrast, It would just stay the same

So it's fixed? A bad contrast value could have been stored in EEPROM making the screen unreadable.

> Are you using the stock Pin out of the board, I have a feeling this is a SKR1.4

Stock SKR 1.4 Turbo, cables (rotated one end of each cable 180 degrees), LCDs. The SKR 1.4 and 1.4 Turbo are the same board except for the MCU (LPC1768 vs. LPC1769).

closed 2021-09-07 by KetamineKyle