MarlinFirmware/Marlin #22722: [BUG] Need sanity check with warning G38_PROBE_TARGET & //USE_ZMIN_PLUG

electronicm, 2021-09-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I use a probe for Z homing. (connected on the BLTouch port of the Creality 4.2.2) 
After homing, software endstops are used and physical endstops are ignored, also for the Z-axis.
Since this makes the Z end stop superfluous, I removed it and planning to use the pin for something else later.

If I now try to implement the whole thing in the FW, I run into some errors when compiling.

```
Compiling .pio\build\STM32F103RET6_creality\src\src\module\settings.cpp.o
In file included from Marlin\src\module\../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\module\../inc/../HAL/HAL.h:30,
                 from Marlin\src\module\../inc/MarlinConfig.h:31,
                 from Marlin\src\module\endstops.h:28,
                 from Marlin\src\module\endstops.cpp:27:
Marlin\src\module\endstops.cpp:944:65: error: 'Z_MIN' was not declared in this scope; did you mean 'Y_MIN'?
  944 |         else if (stepper.axis_is_moving(Z_AXIS)) { _ENDSTOP_HIT(Z, TERN(Z_HOME_TO_MIN, MIN, MAX)); planner.endstop_triggered(Z_AXIS); }
      |                                                                 ^
Marlin\src\module\../inc/../HAL/./STM32/../shared/Marduino.h:42:25: note: in definition of macro '_BV'
   42 | #define _BV(b) (1UL << (b))
      |                         ^
Marlin\src\module\endstops.cpp:863:38: note: in expansion of macro 'SBI'
  863 |   #define _ENDSTOP_HIT(AXIS, MINMAX) SBI(hit_state, _ENDSTOP(AXIS, MINMAX))
      |                                      ^~~
Marlin\src\module\endstops.cpp:607:32: note: in expansion of macro '__ENDSTOP'
  607 | #define _ENDSTOP(AXIS, MINMAX) __ENDSTOP(AXIS, MINMAX)
      |                                ^~~~~~~~~
Marlin\src\module\endstops.cpp:863:53: note: in expansion of macro '_ENDSTOP'
  863 |   #define _ENDSTOP_HIT(AXIS, MINMAX) SBI(hit_state, _ENDSTOP(AXIS, MINMAX))
      |                                                     ^~~~~~~~
Marlin\src\module\endstops.cpp:944:52: note: in expansion of macro '_ENDSTOP_HIT'
  944 |         else if (stepper.axis_is_moving(Z_AXIS)) { _ENDSTOP_HIT(Z, TERN(Z_HOME_TO_MIN, MIN, MAX)); planner.endstop_triggered(Z_AXIS); }
      |                                                    ^~~~~~~~~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\module\endstops.cpp.o] Error 1
```

### Expected behavior

It should work or give a sanity check with warning.

### Actual behavior

I disabled G38_PROBE_TARGET.

### Steps to Reproduce

compile FW with my [Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/7117279/Configurations.zip) or use stock CFG and adjust following:

Marlin\Configuration.h
```// #define USE_ZMIN_PLUG
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
#define BLTOUCH
#define Z_SAFE_HOMING
```
Marlin\Configuration_adv.h
```
#define G38_PROBE_TARGET
```

### Version of Marlin Firmware

bugfix-2.0.x(2021-09-06)

### Additional information & file uploads

Just reported it for completeness.

I don't need G38_PROBE_TARGET at the moment and have disabled it.

thisiskeithb, 2021-09-06:
`HAS_G38_PROBE` seems to have no concept of Z homing without an endstop.

jsrobson10, 2023-04-16:
i can confirm, this is still an issue on version 2.1.2. 

a workaround (or partial solution) is to modify line 1123 in `Marlin/src/module/endstops.cpp`. change
```
#if HAS_Z_AXIS
```
to this
```
#if HAS_Z_AXIS && USE_ZMIN_PLUG
```

closed 2023-04-19 by thinkyhead