MarlinFirmware/Marlin #22723: [BUG] Load / Unload selections from the menu do not use correct extruder when in IDEX duplicate mode

ishapiro, 2021-09-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When the printer is set to an IDEX Mode of duplicate or mirror you are unable to use the LCD load and unload commands because independent of the extruder select the 1st extruder (tool 0) is always used to extrude the filament.

This behavior does not impact the preheat function which functions properly even when in a duplication mode.

When in Auto Park mode the unload and load selections work properly.

### Bug Timeline

Unknown

### Expected behavior

The LCD menu options to load and unload the filament and the options to move the extruder should work properly independent of the IDEX mode or if this is not possible display an error message.

### Actual behavior

The first extruder always moves in response to LCD extruder movement and load/unload commands when in IDEX duplicate or mirror mode.

### Steps to Reproduce

1. Test the proper operation by loading and unloading filament in both of the extruders on an IDEX printer.
2. Set the IDEX mode (X mode) to duplicate.
3. Attempt to load filament into the 2nd extruder.
4. If the first extruder is up to temp it will extruder filament instead.

### Version of Marlin Firmware

V2.3.3 (https://github.com/JGMaker3dofficial/ArtistD-Pro)

### Printer model

JG Maker Artist-D Pro

### Electronics

MKS Robin Pro

thisiskeithb, 2021-09-06:
> Version of Marlin Firmware
> V2.3.3 (https://github.com/JGMaker3dofficial/ArtistD-Pro)

We can't support forks of Marlin here, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

ishapiro, 2021-09-06:
I understand.

Irv

- - - - - - - - -

[image: photo]
*Irv Shapiro | Personal Email*
Cogitations, Inc.

Mobile: 847 331 1310 | ***@***.***

http://www.cogitations.com
<https://mailtrack.io/trace/link/0034abbe04193ab7408827d29e4eac3ffc5414ca?url=http%3A%2F%2Fwww.cogitations.com&userId=2835845&signature=c0c0e6d9172e14a4>
| Skype: ishapiro55 <#SignatureSanitizer_SafeHtmlFilter_>
<https://mailtrack.io/trace/link/22e6f96718a61bc0ba44e04fa8c779f8cc38f235?url=http%3A%2F%2Fus.linkedin.com%2Fin%2Firvshapiro&userId=2835845&signature=a1ce3e1e7bbef143>
<https://mailtrack.io/trace/link/1df40be7d9272139240115063b09488faf1db19b?url=http%3A%2F%2Ftwitter.com%2Firvshapiro&userId=2835845&signature=3c6476d7873f24d9>

thisiskeithb, 2021-09-08:
Issue opened in the fork's repo: https://github.com/JGMaker3dofficial/ArtistD-Pro/issues/3

closed 2021-09-08 by thisiskeithb