MarlinFirmware/Marlin #22725: [BUG] Printing with Jyers UI on Ender3V2 causes firmware crash

123marvin123, 2021-09-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When printing something on the Ender 3 V2 using the recently merged Jyers UI, the printer firmware will crash when the **rolling filename** (which displays the name of the currently printed file) reaches it's last character.

### Bug Timeline

New bug, NOT happening on Jyers fork of Marlin

### Expected behavior

The rolling filename should continue displaying the whole filename and the printer should continue executing GCode.

### Actual behavior

When the last character of the filename (which moves from right to left on the screen) reaches the left side of the screen, the printer / lcd freezes for one second and then reboots.

### Steps to Reproduce

1. Flash firmware with the DWIN_CREALITY_LCD_JYERSUI definition enabled.
2. Print any file with a long filename.
3. Watch the filename moving from right to left until the left character of the filename is at the left edge of the screen.
4. Printer freezes and reboots

### Version of Marlin Firmware

On commit 0f61d9e4dd4d4e4f27e5c688ab2c5dbd0f03af84 of bugfix 2.0.x branch

### Printer model

Creality Ender 3 v2

### Electronics

Stock board

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

The issue occurs in the `CrealityDWINClass::Draw_Print_Filename` function and I want to emphasize that this issue does not occur in the Marlin fork by Jyers. I copied over the Draw_Print_Filename function and the problem is fixed, so there must be something going on inside of this function which causes the crash.

123marvin123, 2021-09-07:
@Jyers

Jyers, 2021-09-07:
I had a look at the changes to the function. They all mostly seem to be formatting and cleanup, but if you've tested and found that my function works when copied into the upstream code, then my best guess would be some error caused by `len = _MIN(pos, 30);`

thinkyhead, 2021-09-13:
Marlin uses a well-tested method for displaying a scrolling status message which is a good guide on how to draw a continuously scrolling (and repeating) message, such as a long filename. I'd recommend re-implementing `Draw_Print_Filename` following the methodology in the code associated with `STATUS_MESSAGE_SCROLLING` — which also properly counts the number of `UTF-8` characters and advances the scroll the appropriate number of bytes when dealing with UTF-8 characters.

Jyers, 2021-09-19:
I'm currently getting my fork setup with a fresh branch based on the latest bugfix so I can easily PR fixes and features. Once I have everything sorted I can start working to get some PRs going.

tome9111991, 2021-09-21:
I can confirm 
Display firmware crash with long filename.

closed 2021-12-01 by github-actions[bot]