MarlinFirmware/Marlin #22726: [FR] button multiplexing

bulbersator, 2021-09-07:
### Is your feature request related to a problem? Please describe.

Button support has been added recently, but most motherboards don't have many buttons free. Multiplexing will allow more buttons to be operated.

### Describe the feature you want

In the button configuration, I would like to be able to set the pins that support the buttons and their behavior.
Like in the table below with pin names and button behaviour.

[image]

thinkyhead, 2021-09-13:
It's a pretty neat idea and not very difficult. Someone will likely take an interest and implement it.

UMO-printing, 2021-12-22:
Arduino has a library for reading matrix buttons, perhaps that would be a good point to start from. 
The Arduino library to start from is here: https://github.com/Nullkraft/Keypad

The original creators of this library are: Mark Stanley and Alexander Brevig 
The official Arduino library is owned by Alexander Brevig, but he has not been contactable for several years, so Mark Stanley has been the one who has been updating a Keypad library in recent years. And Mark Stanley's updated library is available here: https://github.com/Nullkraft/Keypad

Among other functions, the Nullkraft/Keypad library can deal with multiple, simultaneous, keypresses.

SauravMaheshkar, 2022-01-01:
I'd like to take this up @thinkyhead, can anyone guide as to where the changes should be made ?

UMO-printing, 2022-01-01:
Hmmm...  an important part of the "challenge" is: to determine where, how and what changes should be made. 

Whenever I try to implement an idea, I usually start by searching for the keywords used in the `#define` statements in the `config.h` and/or `config_adv.h` files. 
Such searches will uvenile all the places in the code, where these keywords are referenced. 

From there, one can try to uncover what the current code does. 
And once you know that, then it is time to think up how to implement the new code. 

Well, these are the steps I tend to use. Others might use a different approach.

UMO-printing, 2022-01-01:
It would seem that custom buttons are set in `config_adv.h` 
using the `#define CUSTOM_USER_BUTTONS` token

And it would seem, that much program action about custom buttons happens in `MarlinCore.cpp`

Does this help you to see where the current code is?

bulbersator, 2022-01-01:
I started working on this, but in the meantime I switched to other firmware
for other reasons. That's fairly easy to implement, you just need to use
the existing custom user buttons function and simply add logic conditions
for checking 3 or more pins, instead of one.