MarlinFirmware/Marlin #22729: [BUG] Marlin will not compile with sensorless probing and z_min disabled

spotmaxdog, 2021-09-08:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I am trying to compile marlin for my QQ-S Pro with an SKR 2 board and TMC2209 steppers.

For sensorless probing, I am enabling SENSORLESS_PROBING and defining my drivers as TMC2209s.

I am commenting out my Z_min endstop (old level sensor) as I want to use the 2209s to probe.

When I go to compile I get this error. I can get it to compile if I define an nonexistent Z_Min endstop but wouldn't that make sensorless probing fail?

[image]

### Expected behavior

Expected to compile with no Z_min endstop, using sensorless probing

### Actual behavior

Compiling crashed.

Can get it to compile if I go into endstops.h and comment out the line but I do not know what that does to Marlin

### Steps to Reproduce

Enable SENSORLESS_PROBING
Disable all Z_Min endstops
Compile

### Version of Marlin Firmware

2.0.9.1

### Printer model

QQ-S Pro

### Electronics

BTT SKR 2.0 with TMC 2209s

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7126290/Archive.zip)

thinkyhead, 2021-09-08:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2021-09-11:
This seems to be solved if you enable `USE_ZMIN_PLUG`. If that works for you then this can be closed.

thinkyhead, 2021-09-11:
I found a more proper solution, and will push a patch shortly. Please test the `bugfix-2.0.x` branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we can investigate further.

thinkyhead, 2021-09-11:
You may want to enable `HOST_ACTION_COMMANDS` when updating your configurations.

commit 011329fe (2021-09-11): 🐛 No probe enum for DELTA + SENSORLESS_PROBING

Fix #22729
closed 2021-09-14 by thinkyhead