MarlinFirmware/Marlin #22731: [BUG] (G29 only probes first point and then returns [ERROR] Error:Probing Failed)

smurfster, 2021-09-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a Creality Ender 3 with a V4.2.2 board.
So ive been building a DIY ABL https://www.thingiverse.com/thing:3293821

I have compiled the latest stable release (2.0.9.1) and the bugfix release both have the same issue.

I have attached the config files

The probe can be operated with all relavent commands. G28 works no problems, But when I execute G29 is attempts the first position that returns Error:Probing Failed even though it looked like it worked

I enabled DEBUG_LEVELING_FEATURE and enabled it using M111 S32. Then executed G28 and G29

>>> M111 S32
SENDING:M111 S32
echo:DEBUG:LEVELING

>>> G28
SENDING:G28
>>> G28  X53.00 Y30.00 Z6.00
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X-43.00 Y-20.00 Z0.00 (Left-Front & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
Raise Z (before homing) by 2.00
>>> do_blocking_move_to  X53.00 Y30.00 Z6.00
>  X53.00 Y30.00 Z6.00
<<< do_blocking_move_to  X53.00 Y30.00 Z6.00
  current_position= X0.00 Y0.00 Z6.00 : sync_plan_position
>>> do_blocking_move_to  X0.00 Y0.00 Z6.00
>  X-352.50 Y-352.50 Z6.00
<<< do_blocking_move_to  X-352.50 Y-352.50 Z6.00
>>> homeaxis(X)
Home Fast: -352.50mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(X, -352.50, [20.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z6.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(X, 5.00, [20.00])
<<< do_homing_move  X0.00 Y0.00 Z6.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(X, -10.00, 10.00)
<<< do_homing_move  X0.00 Y0.00 Z6.00
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z6.00 :
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z6.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z6.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home Fast: -352.50mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(Y, -352.50, [20.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z6.00
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(Y, 5.00, [20.00])
<<< do_homing_move  X0.00 Y0.00 Z6.00
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z6.00
...(Y, -10.00, 10.00)
<<< do_homing_move  X0.00 Y0.00 Z6.00
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X0.00 Y0.00 Z6.00 :
<<< set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z6.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z6.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
>>> home_z_safely  X0.00 Y0.00 Z6.00
  current_position= X0.00 Y0.00 Z6.00 : sync_plan_position
  destination= X160.00 Y137.00 Z0.00 : home_z_safely
>>> do_blocking_move_to  X0.00 Y0.00 Z6.00
>  X160.00 Y137.00 Z6.00
echo:busy: processing
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
>>> homeaxis(Z)
  current_position= X160.00 Y137.00 Z6.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(6.00)
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X160.00 Y137.00 Z6.00
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X160.00 Y137.00 Z6.00
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
Home Fast: -375.00mm
>>> do_homing_move  X160.00 Y137.00 Z6.00
...(Z, -375.00, [4.00])
echo:busy: processing
<<< do_homing_move  X160.00 Y137.00 Z6.00
Move Away: 6.00mm
>>> do_homing_move  X160.00 Y137.00 Z6.00
...(Z, 6.00, 4.00)
echo:busy: processing
<<< do_homing_move  X160.00 Y137.00 Z6.00
Re-bump: -12.00mm
>>> do_homing_move  X160.00 Y137.00 Z6.00
...(Z, -12.00, 2.00)
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X160.00 Y137.00 Z6.00
>>> set_axis_is_at_home(Z)
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> probe.offset.z = 0.00
Axis Z home_offset = 0.00 position_shift = 0.00
> home_offset[Z] = 0.00
  current_position= X160.00 Y137.00 Z0.00 :
<<< set_axis_is_at_home(Z)
  current_position= X160.00 Y137.00 Z0.00 : sync_plan_position
  current_position= X160.00 Y137.00 Z0.00 : > AFTER set_axis_is_at_home
  current_position= X160.00 Y137.00 Z0.00 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(6.00)
>>> do_blocking_move_to  X160.00 Y137.00 Z0.00
>  X160.00 Y137.00 Z6.00
echo:busy: processing
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X160.00 Y137.00 Z6.00
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
<<< homeaxis(Z)
<<< home_z_safely  X160.00 Y137.00 Z6.00
  current_position= X160.00 Y137.00 Z6.00 : sync_plan_position
<<< G28  X160.00 Y137.00 Z6.00

>>> G29
SENDING:G29
>>> G29  X160.00 Y137.00 Z6.00
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X-43.00 Y-20.00 Z0.00 (Left-Front & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position= X160.00 Y137.00 Z6.00 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(6.00)
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X160.00 Y137.00 Z6.00
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X160.00 Y137.00 Z6.00
<<< do_blocking_move_to  X160.00 Y137.00 Z6.00
>>> Probe::probe_at_point  X160.00 Y137.00 Z6.00
...(10.00, 10.00, raise, 0, probe_relative)
  current_position= X160.00 Y137.00 Z6.00 :
>>> do_blocking_move_to  X160.00 Y137.00 Z6.00
>  X53.00 Y30.00 Z6.00
echo:busy: processing
<<< do_blocking_move_to  X53.00 Y30.00 Z6.00
  current_position= X53.00 Y30.00 Z6.00 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X53.00 Y30.00 Z6.00
>>> Probe::probe_down_to_z  X53.00 Y30.00 Z6.00
>>> do_blocking_move_to  X53.00 Y30.00 Z6.00
>  X53.00 Y30.00 Z-2.00
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to  X53.00 Y30.00 Z-2.00
  current_position= X53.00 Y30.00 Z-2.00 : sync_plan_position
<<< Probe::probe_down_to_z  X53.00 Y30.00 Z-2.00
SLOW Probe fail! - No trigger.
<<< Probe::run_z_probe  X53.00 Y30.00 Z-2.00
  current_position= X53.00 Y30.00 Z-2.00 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(6.00)
>>> do_blocking_move_to  X53.00 Y30.00 Z-2.00
>  X53.00 Y30.00 Z6.00
echo:busy: processing
<<< do_blocking_move_to  X53.00 Y30.00 Z6.00
>>> do_blocking_move_to  X53.00 Y30.00 Z6.00
>  X53.00 Y30.00 Z6.00
<<< do_blocking_move_to  X53.00 Y30.00 Z6.00
Error:Probing Failed
[ERROR] Error:Probing Failed

<<< Probe::probe_at_point  X53.00 Y30.00 Z6.00
  current_position= X53.00 Y30.00 Z6.00 : Probe::set_deployed
deploy: 0
  current_position= X53.00 Y30.00 Z6.00 : > probing complete
<<< G29  X53.00 Y30.00 Z6.00

### Expected behavior

I expected to measure the bed level at muitiple points

### Actual behavior

Attempted bed level check at the first point then displayed probe failed

### Steps to Reproduce

Use command G28 then G29

### Version of Marlin Firmware

2.0.9.1 and Marlin bugfix-2.0.x (Sep  8 2021 09:02:00) 

### Printer model

Creality Ender 3

### Electronics

Stock with v4.2.2 board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7129507/config.zip)

thinkyhead, 2021-09-11:
Try disabling `Z_PROBE_LOW_POINT` or setting it to a larger negative value like -10.

smurfster, 2021-09-13:
> Try disabling `Z_PROBE_LOW_POINT` or setting it to a larger negative value like -10.

awesome that fixed it, thank you

closed 2021-09-13 by smurfster