MarlinFirmware/Marlin #22732: [FR] send G29 Messages as //action:notification

electronicm, 2021-09-08:
### Describe the feature you want

When I run a G29, the printer display shows the message `Probing mesh point X/X`. This message is also sent to the host, but not as `//action:notification`.
So it will not be interpreted as a notification, e.g. from OctoPrint, and listed in the appropriate place.
Could these status messages be sent as //action:notification?

If there is already another solution to get e.g. the messages from the G29 into the printer notification area without much effort, I appreciate any suggestion.

thinkyhead, 2021-09-13:
The printing of the mesh leveling progress to serial is kind of hack-ish to begin with, but OctoPrint is accustomed to intercepting and interpreting serial output from the firmware to display the information … someplace. As I understand it, a 'notification' puts up a dialog box with an "Okay" button, so I don't know if that is the appropriate method for this message. Maybe it's enough to preface it with "echo:" as you can test with this change (using `bugfix-2.0.x`):
```diff
- SERIAL_ECHOLNPGM("Probing mesh point ", point_num, "/", GRID_MAX_POINTS, ".");
+ SERIAL_ECHO_MSG("Probing mesh point ", point_num, "/", GRID_MAX_POINTS, ".");
```

cp2004, 2021-09-13:
Using `action: notification` means that the message is added to a section in the sidebar of OctoPrint, or optionally as a popup notification. As far as I knew it was only the host prompts that created dialog boxes.

closed 2025-06-22 by electronicm