MarlinFirmware/Marlin #22735: [BUG] Compile Error with DWIN_CREALITY_LCD

tome9111991, 2021-09-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello
i have a Ender 3v2 with SKR E3 Turbo and stock E3v2 Display.
with the latest source 

i iuse `#define DWIN_CREALITY_LCD`

i got a error durring build the .bin file

```
1769/src/src/feature/pause.cpp.o: in function `load_filament(float, float, float, signed char, bool, bool, PauseMode)':
C:\Users\username\3D Druck\Ender 3v2\Marlin-bugfix-2.0.x/Marlin\src\feature/pause.cpp:184: undefined reference to `MarlinUI::pause_show_message(PauseMessage, PauseMode, unsigned char)'
c:/users/username/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\username\3D Druck\Ender 3v2\Marlin-bugfix-2.0.x/Marlin\src\feature/pause.cpp:189: undefined reference to `MarlinUI::pause_show_message(PauseMessage, PauseMode, unsigned char)'
c:/users/username/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\username\3D Druck\Ender 3v2\Marlin-bugfix-2.0.x/Marlin\src\feature/pause.cpp:210: undefined reference to `MarlinUI::pause_show_message(PauseMessage, PauseMode, unsigned char)'
c:/users/username/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\username\3D Druck\Ender 3v2\Marlin-bugfix-2.0.x/Marlin\src\feature/pause.cpp:258: undefined reference to `MarlinUI::pause_show_message(PauseMessage, PauseMode, unsigned char)'
c:/users/username/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/feature/pause.cpp.o:C:\Users\username\3D Druck\Ender 3v2\Marlin-bugfix-2.0.x/Marlin\src\feature/pause.cpp:329: more undefined references to `MarlinUI::pause_show_message(PauseMessage, PauseMode, unsigned char)' follow
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
```

Configs:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7138332/Marlin.zip)

### Version of Marlin Firmware

bugfix-2.0.x

ellensp, 2021-09-09:
you can get this to compile by adding this block to Marlin/src/lcd/e3v2/creality/dwin.cpp
```
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  void MarlinUI::pause_show_message(const PauseMessage message, const PauseMode mode/*=PAUSE_MODE_SAME*/, const uint8_t extruder/*=active_extruder*/) {
    switch (message) {
      default: break;
    }
  }
#endif
```

But its just s stub, it doesn't do anything

tome9111991, 2021-09-10:
thanks compiling works so far.

commit 453e6095 (2021-09-11): 🐛 Followup to JyersUI

Fix #22735, #22736
closed 2021-09-11 by thinkyhead