MarlinFirmware/Marlin #22736: [BUG] Current Bugfix-2.0.x DWIN_CREALITY_LCD_JYERSUI does not compile missing includes and default config entries

GhostlyCrowd, 2021-09-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Current bugfix 2.0.x 754b3191

https://github.com/MarlinFirmware/Marlin/pull/22422

**Fails to compile,** without adding `#include "../../../gcode/gcode.h"` to Marlin\src\lcd\e3v2\jyersui\dwin.cpp

with errors "Marlin\src\lcd\e3v2\jyersui\dwin.cpp:335:11: error: 'gcode' was not declared in this scope"

**Fails to compile** without enabling `#define POWER_LOSS_RECOVERY` in configuration_adv.h 

with errors `Marlin\src\lcd\e3v2\jyersui\dwin.cpp:4758:11: error: 'recovery' was not declared in this scope; did you mean 'remove'?`

**Fails to compile** because default config/examples missing  `#define ENCODER_5X_STEPS_PER_SEC    30  // (steps/s) Encoder rate for 5x speed` in configuration_adv.h

@jyer

### Bug Timeline

When DWIN_CREALITY_LCD_JYERSUI was presented as an option

### Expected behavior

To compile

### Actual behavior

Fails with the above explanations of errors and how to resolve them

### Steps to Reproduce

Try to compile as the code exists.

### Version of Marlin Firmware

Bugfix 2.0.x

### Electronics

SKR E3 Mini V1.2

electronicm, 2021-09-09:
I've known the problem since DWIN MARLINUI.

GhostlyCrowd, 2021-09-09:
> I've known the problem since DWIN MARLINUI.
Really, I could have sworn I tried that, and it compiled fine today, but DWIN_CREALITY_LCD_ENHANCED also suffers from the same related errors its seems. Just tested,

tome9111991, 2021-09-09:
You use latest config?

https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2/CrealityUI

GhostlyCrowd, 2021-09-09:
> You use latest config?
> 
> https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20V2/CrealityUI

No I use a custom config based on the ender3 skr e3 mini from marlins config example repo and then i use the default config to compare and verify things that are missing.

tome9111991, 2021-09-09:
Maybe you should compare your config with latrst e3v2 config

GhostlyCrowd, 2021-09-09:
I've tracked it down, `DWIN_CREALITY_LCD_JYERSUI` depends on `#define POWER_LOSS_RECOVERY` always being enabled because it is the only place `DWIN_CREALITY_LCD_JYERSUI` seems to get `#include "../../../gcode/gcode.h"` included, from `src_filter=+<src/feature/powerloss.cpp>`

If one doesn't want to run  Power loss recovery, `Marlin\src\lcd\e3v2\jyersui\dwin.cpp` should definitely include `#include "../../../gcode/gcode.h"` rather than hoping it's defined somewhere else.

thinkyhead, 2021-09-11:
Please include configurations when posting bug reports so that we can use them to reproduce the exact conditions for the reported error.

thinkyhead, 2021-09-11:
67d82ff228

tome9111991, 2021-09-11:
> I've tracked it down, `DWIN_CREALITY_LCD_JYERSUI` depends on `#define POWER_LOSS_RECOVERY` always being enabled because it is the only place `DWIN_CREALITY_LCD_JYERSUI` seems to get `#include "../../../gcode/gcode.h"` included, from `src_filter=+<src/feature/powerloss.cpp>`
> 
> If one doesn't want to run Power loss recovery, `Marlin\src\lcd\e3v2\jyersui\dwin.cpp` should definitely include `#include "../../../gcode/gcode.h"` rather than hoping it's defined somewhere else.
> 
> Adding `#include "../../../gcode/gcode.h"` to dwin.cpp squashes several dozen errors related to it not being included with `#define POWER_LOSS_RECOVERY` but then I still get errors because of course `#define POWER_LOSS_RECOVERY` `Marlin\src\lcd\e3v2\jyersui\dwin.cpp:4758:11: error: 'recovery' was not declared in this scope; did you mean 'remove'?`
> 
> If it's somehow required explicitly for this, then a sanity check would be beneficial.

You should use '#define POWER_LOSS_RECOVERY'
You can Power Loss disable with Display GUI there is a checkbox to enable/disable

(In my mind it is disable by default)

GhostlyCrowd, 2021-09-11:
> You should use `#define POWER_LOSS_RECOVERY`
> You can Power Loss disable with Display GUI there is a checkbox to enable/disable
>
> (In my mind it is disable by default)

It's wasted flash space if one is not going to use it, IE: never print from SD and or has a power loss solution already such as a Battery backup and a generator that auto kicks on like i do.

GhostlyCrowd, 2021-09-11:
> [67d82ff](https://github.com/MarlinFirmware/Marlin/commit/67d82ff228789408f7f32e6c9af3108c989c0c90) [453e609](https://github.com/MarlinFirmware/Marlin/commit/453e60958a8cde6d7541b4dc3a99d054b53eb695)
> 
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I can confirm it now compiles and works with out `#define POWER_LOSS_RECOVERY`

commit 669d1fc1 (2021-09-11): 🐛 Followup to JyersUI

Fix #22736
commit ecae5bab (2021-09-11): 🐛 Followup to JyersUI

Fix #22736
closed 2021-09-11 by GhostlyCrowd