MarlinFirmware/Marlin #22741: [BUG] Error on compile with DWIN_CREALITY_LCD_ENHANCED and any AUTO_BED_LEVELING_

electronicm, 2021-09-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I use the current version bugfix-2.0.x with the options
- DWIN_CREALITY_LCD_ENHANCED and
- AUTO_BED_LEVELING_UBL
I get the following error:

```
In file included from Marlin\src\gcode\bedlevel\ubl\../../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\gcode\bedlevel\ubl\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\gcode\bedlevel\ubl\M421.cpp:27:
Marlin\src\gcode\bedlevel\ubl\M421.cpp: In static member function 'static void GcodeSuite::M421()':
Marlin\src\gcode\bedlevel\ubl\M421.cpp:72:39: error: 'DWIN_MeshUpdate' was not declared in this scope; did you mean 'DWIN_Update'?
   72 |     TERN_(DWIN_CREALITY_LCD_ENHANCED, DWIN_MeshUpdate(ij.x, ij.y, zval));
      |                                       ^~~~~~~~~~~~~~~
Marlin\src\gcode\bedlevel\ubl\../../../inc/../core/macros.h:560:26: note: in definition of macro 'THIRD'
  560 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\gcode\bedlevel\ubl\../../../inc/../core/macros.h:204:29: note: in expansion of macro '___TERN'
  204 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\gcode\bedlevel\ubl\../../../inc/../core/macros.h:203:29: note: in expansion of macro '__TERN'
  203 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\gcode\bedlevel\ubl\../../../inc/../core/macros.h:202:29: note: in expansion of macro '_TERN'
  202 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin\src\gcode\bedlevel\ubl\M421.cpp:72:5: note: in expansion of macro 'TERN_'
   72 |     TERN_(DWIN_CREALITY_LCD_ENHANCED, DWIN_MeshUpdate(ij.x, ij.y, zval));
      |     ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\gcode\bedlevel\ubl\M421.cpp.o] Error 1
```

If I use the current version bugfix-2.0.x with the options
- DWIN_CREALITY_LCD_ENHANCED and
- AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_LINEAR or AUTO_BED_LEVELING_3POINT
I get the following error:
```
Marlin\src\lcd\e3v2\enhanced\dwin.cpp: In function 'void DWIN_Progress_Update()':
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:1802:7: error: 'parser' was not declared in this scope
 1802 |   if (parser.seenval('P')) _percent_done = parser.byteval('P');
      |       ^~~~~~
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:1803:7: error: 'parser' was not declared in this scope
 1803 |   if (parser.seenval('R')) _remain_time = parser.ulongval('R') * 60;
      |       ^~~~~~
Marlin\src\lcd\e3v2\enhanced\dwin.cpp: In function 'void SetMoveZto0()':
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:2179:3: error: 'gcode' was not declared in this scope
 2179 |   gcode.process_subcommands_now_P(cmd);
      |   ^~~~~
Marlin\src\lcd\e3v2\enhanced\dwin.cpp: In function 'void SetPID(celsius_t, heater_id_t)':
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:2192:3: error: 'gcode' was not declared in this scope
 2192 |   gcode.process_subcommands_now_P(cmd);
      |   ^~~~~
Marlin\src\lcd\e3v2\enhanced\dwin.cpp: In function 'void SetPwrLossr()':
Marlin\src\lcd\e3v2\enhanced\dwin.cpp:2200:3: error: 'recovery' was not declared in this scope; did you mean 'remove'?
 2200 |   recovery.enable(!recovery.enabled);
      |   ^~~~~~~~
      |   remove
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\enhanced\dwin.cpp.o] Error 1
```

### Bug Timeline

new bug(?)

### Expected behavior

It should work or give a sanity check with warning.

### Actual behavior

I do not try DWIN_CREALITY_LCD_ENHANCED.

### Steps to Reproduce

Compile the mentioned version with the described customizations and my [configs](https://github.com/MarlinFirmware/Marlin/files/7144437/Configuration.zip).

### Version of Marlin Firmware

bugfix-2.0.x(2021-09-10)

### Printer model

Creality Ender 3v2

### Electronics

Creality 4.2.2

### Add-ons

CR-Touch

### Bed Leveling

UBL Bilinear mesh

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7144437/Configuration.zip)

thinkyhead, 2021-09-11:
These are relatively simple to patch. I'll push a fix shortly.

CC: @mriscoc

closed 2021-09-11 by thinkyhead

mriscoc, 2021-09-14:
> These are relatively simple to patch. I'll push a fix shortly.
> 
> CC: @mriscoc

Thanks

commit 953d283e (2021-09-11): 🐛 Followup to E3V2 Enhanced

Fix #22741