MarlinFirmware/Marlin #22744: compile error in the original marlin version on line 193ff of file  ui_api.h

awiltschek, 2021-09-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

On a compiled version from digant arduion raises an error that ui.lcd_clicked is a read only !
In the original Marlin from you this statement is #if defines out.

**Version of Marlin compiled by User Digant:**
// inline void simulateUserClick() { ui.lcd_clicked = true; }

**your original Marlin version:**
 inline void simulateUserClick() {
    #if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
      ui.lcd_clicked = true;
    #endif
  }

if ui.lcd_clicked is a readonly then the bug is allready existing in the original marling version as you must not initalitze it with true.
you can find the error easily by remarking the #if / #endif lines on line 193ff.

i tested it and copied your original code into the version of digant and it compiled without error.
i remarked the #if / #endif and the error was back again.

so in case one of the #if either ... is true you will raise this compile error - so this cannot be correct code or nether tested.

**Error:**
In file included from C:\Users\awilt\Desktop\BLTouch\Marlin-2.0.x\Marlin\src\lcd\extui\ftdi_eve_touch_ui\compat.h:31:0,
                 from C:\Users\awilt\Desktop\BLTouch\Marlin-2.0.x\Marlin\src\lcd\extui\ftdi_eve_touch_ui\archim2-flash\flash_storage.cpp:23:
C:\Users\awilt\Desktop\BLTouch\Marlin-2.0.x\Marlin\src\lcd\extui\ui_api.h: In function 'void ExtUI::simulateUserClick()':
C:\Users\awilt\Desktop\BLTouch\Marlin-2.0.x\Marlin\src\lcd\extui\ui_api.h:197:24: error: assignment of read-only variable 'MarlinUI::lcd_clicked'
       ui.lcd_clicked = true;
                        ^~~~

### Steps to Reproduce

1) go into file ui_api.h of dir "Marlin-2.0.x\Marlin\src\lcd\extui"
2) remark lines 198 and 200 so the code looks like this

**inline void simulateUserClick() {
 //    #if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
      ui.lcd_clicked = true;
//     #endif
  }**

3) compile and fix the error so this is not happening again :-)

### Version of Marlin Firmware

2.0.9.1

### Printer model

Sidewinder X1

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

Other (explain below)

### Additional information & file uploads

none all said

awiltschek, 2021-09-10:
You just need to remove the init by true

inline void simulateUserClick() {
 //    #if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
      ui.lcd_clicked;
 //    #endif
  }

awiltschek, 2021-09-10:
it seems to be a mess of arduino gui, as lcd_clicked is a normal variable (not read only as the error says).

#if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
  bool MarlinUI::lcd_clicked;
#endif

thisiskeithb, 2021-09-10:
The code you're using is from a fork and out of date.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-10 by thisiskeithb

awiltschek, 2021-09-10:
i think here is the trouble with the arduino gui why it raises the compiling error 
...\Desktop\BLTouch\Marlin-2.0.x\Marlin\src\lcd\extui\ui_api.h:197:24: error: assignment of read-only variable 'MarlinUI::lcd_clicked'
ui.lcd_clicked = true;

Marlinui.h - line 543ff:

    static **constexpr** bool lcd_clicked = false;

#if EITHER(HAS_LCD_MENU, EXTENSIBLE_UI)
    static bool lcd_clicked;
    static inline bool use_click() {
      const bool click = lcd_clicked;
      lcd_clicked = false;
      return click;
    }
  #else
    static constexpr bool lcd_clicked = false;
    static inline bool use_click() { return false; }
  #endif

awiltschek, 2021-09-10:
> The code you're using is from a fork and out of date.
>
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted [...]

interesting - i downloaded latest version from Marlin Homepage and the code and error is the same :-)
look on line 543 in marlinui.h the same "static constexpr bool lcd_clicked = false;"

awiltschek, 2021-09-10:
the trouble is with the version from user diant who removed the #if / #endif statement which makes sense when you use it as its shown in the code ! so no need for a bug fix.