MarlinFirmware/Marlin #22747: [BUG] independent DEFAULT_AXIS_STEPS_PER_UNIT for X2 Carriage

mindfreak269, 2021-09-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

how to setup independent DEFAULT_AXIS_STEPS_PER_UNIT for X2 ? i have used BTT S42B for X1 carriage and normal open loop stepper for X2, BTT S42B have 102.4 step /mm for 16 microstepping and normal openloop user 80 step/mm for 16 microstepping, since X1 and X2 use same step/mm configuration, X1 get inccorect move.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Custom IDEX

### Electronics

BTT SKR 1.4 Turbo

### Add-ons

Expansion Motor Module

thisiskeithb, 2021-09-11:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22740

thisiskeithb, 2021-09-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-11 by thisiskeithb

ellensp, 2021-09-11:
You need identical steppers ie identical steps/mm .  In IDEX the step macro is change to toggle two pins instead of one.   There no way you can use different steps/mm. Its not possible without a serious re-write of marlin.

mindfreak269, 2021-09-11:
> 
> 
> You need identical steppers ie identical steps/mm . In IDEX the step macro is change to toggle two pins instead of one. There no way you can use different steps/mm. Its not possible without a serious re-write of marlin.

yes, i have tried to add mondification but its take big change and i dont want do it . for while i will exchange X2 with Y closed loop motor for while. thanks for advice.