MarlinFirmware/Marlin #22748: [BUG] Linking error while compiling the latest bugfix-2.0.x for MKS_Robin_Nano

bkiziuk, 2021-09-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While linking, the following error occurs:
```Linking .pio/build/mks_robin_nano35/firmware.elf
/home/kiziuk/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file common.inc: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/mks_robin_nano35/firmware.elf] Error 1```

### Bug Timeline

The last bugfix-2.0.x I used was 2 months old. Today, I updated to the latest upstream branch and tried to compile it. 

### Expected behavior

The linking should succeed.

### Actual behavior

While linking, the following error occurs:
```Linking .pio/build/mks_robin_nano35/firmware.elf
/home/kiziuk/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file common.inc: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/mks_robin_nano35/firmware.elf] Error 1```

[bugfix-2.0.x_20210910_linking-issue.zip](https://github.com/MarlinFirmware/Marlin/files/7147411/bugfix-2.0.x_20210910_linking-issue.zip)

### Version of Marlin Firmware

bugfix-2.0.x as of today

### Printer model

Sapphire Pro

### Electronics

MKS Robin Nano 1.2

### Add-ons

MKS_PWC

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

ellensp, 2021-09-11:
That often means there was an error further up, scroll up the build log to see the first error

ellensp, 2021-09-11:
from your zip, which is linked in a weird way so I didn't notice it...
"arm-none-eabi/bin/ld: cannot open linker script file common.inc: No such file"  this is odd

ellensp, 2021-09-11:
I get a number of undefined pin errors...  from provided configs
Setting //#define PSU_CONTROL and #define TEMP_SENSOR_CHAMBER 0 and this compiles without issues.

delete your Marlin/.pio directory and try again.

ellensp, 2021-09-11:
adding the following with pins chosen at random to your Config files, also compiles without issues
#define TEMP_CHAMBER_PIN 5
#define PS_ON_PIN 6

bkiziuk, 2021-09-11:
> from your zip, which is linked in a weird way so I didn't notice it...

I just filled a new form for bug report. Maybe the parsing went wrong...

> delete your Marlin/.pio directory and try again.

Ok, Let me look into this again. There was similar issue reported some time ago [#16450](https://github.com/MarlinFirmware/Marlin/issues/16450)

bkiziuk, 2021-09-11:
> delete your Marlin/.pio directory and try again

Did not help.
I tried to track down the pipeline of scripts but it was beyond my abilities.

At the next iteration, I wiped-out the entire PlatformIO including 
`./Marlin/.pio` and `/home/kiziuk/.platformio` directories and started from scratch.
Now, I have missing variant.h file, much the same as in [#22146](https://github.com/MarlinFirmware/Marlin/issues/22146)

[bugfix-2.0.x_20210910_variant.h-issue.txt](https://github.com/MarlinFirmware/Marlin/files/7147586/bugfix-2.0.x_20210910_variant.h-issue.txt)

ellensp, 2021-09-11:
Did the same, removed everything. It compiled  without issues 

Check the following is in your copy.
file ini/stm32f1.ini  has
[env:mks_robin_nano35]
platform                    = ${common_stm32.platform}
extends                     = stm32_variant
board                       = genericSTM32F103VE
board_build.variant         = MARLIN_F103Vx
...

file ini/stm32-common.ini has
[stm32_variant]
extends       = common_stm32
extra_scripts = ${common_stm32.extra_scripts}
                pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
...

the script buildroot/share/PlatformIO/scripts/generic_create_variant.py  copies the variant directory from 

buildroot/share/PlatformIO/variants/MARLIN_F103Vx       (the folder comes from "board_build.variant  = MARLIN_F103Vx"

to

.platformio/packages/framework-arduinoststm32/variants

Where you should find a MARLIN_F103Vx directory.

closed 2021-09-11 by thisiskeithb

bkiziuk, 2021-09-12:
So, for me and the bugfix-2.0.x branch, the script indeed does not copy `buildroot/share/PlatformIO/variants/MARLIN_F103Vx` to the target location - that was the cause of missing `variant.h`.
However I still can't figure out where and which the `common.inc` linker script should be.

ellensp, 2021-09-12:
common.inc is not required for your controller. It doesn't exists for that variant.

It is also not mentioned in your bugfix-2.0.x_20210910_variant.h-issue.txt file

My guess is you have a broken/old python installed that is making a right mess of things.
Uninstall python and install a current version

bkiziuk, 2021-09-14:
Well, something must be fundamentally wrong then. I took a fresh virtual machine with current Fedora Workstation, installed VSCode, then installed PlatformIO from VSCode's extensions. I cloned the bugfix-2.0.x branch, put in my config files.
I got variant.h error at the start. After manually fixing it, I still get linking error at the end. 
I repeated it with the fresh Mint installation - so my guess is that the current version of PlatformIO, the tool-chain or the Python is somehow messing it around.

ellensp, 2021-09-14:
I just replicated your test virtual machine. Marlin still compiled without issues

I setup Virtualbox, downloaded Fedora-Workstation-Live-x86_64-34-1.2.iso and created a virtual machine
15gig drive, 4 cores, 2048MB ram 
Downloaded and installed vscode following this https://computingforgeeks.com/install-visual-studio-code-on-fedora/
Installed platformio plugin into vscode
Git cloned https://github.com/MarlinFirmware/Marlin.git  
Downloaded your configs and copied them into marlin

Hit Build ,  It errored not liking your config serial numbers (bugfix serial number just increased by 1 today) and needing the additional pins.
Updated those simple config file changes, and it compiled fine.

bkiziuk, 2021-09-14:
No idea what is happening, I'm sorry. I already thought there's nothing that can be specific to my testing. I'll try to look at it again...

ellensp, 2021-09-14:
I've just replicated your common.inc issue!!!
Your platfomio.ini file is a old, pre split up into multiple ini files version
I never used this file. I just copied the "default_envs = mks_robin_nano35" line, as that is al you need to set in this file.

Your platformio.ini has it own [env:mks_robin_nano35]  section that breaks the one in ini/stm32f1.ini
Delete that platformio.ini file immediately!

Use the correct one for modern marlin https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.0.x/platformio.ini

All you need to change in platformio.ini is the default environment to "default_envs = mks_robin_nano35"

Do not copy old platfomio.ini files from anywhere, ever.

bkiziuk, 2021-09-14:
I confirm that replacing platformio.ini solved the problem. Thank you. Lesson well learned.