MarlinFirmware/Marlin #22750: [FR] Change GitHub description

bryanpedini, 2021-09-11:
### Describe the feature you want

Now that SMT32 boards are widespread and happily supported, it would be nice/wise to change the GitHub description box to either remove the mention to the Arduino platform, or include 32 bit boards as well.

ellensp, 2021-09-12:
They all still use Arduino framework

thinkyhead, 2021-09-13:
Indeed. The name "Arduino" in that case refers to the API which is available for all kinds of boards. Any board with an Arduino API implementation should be able to run Marlin if a HAL is created for it.

closed 2021-09-13 by thinkyhead

bryanpedini, 2021-09-13:
Got it, so that includes STM32 boards.
I thought that Arduino, of course, meant the Arduino platform itself. Glad to see I was wrong because I learned something new.