MarlinFirmware/Marlin #22751: [BUG] IDEX problem with double distance travel of X2 compared to X1 (Marlin 2.0.9.1; BTT Octopus; TMC2209)

MicSchi, 2021-09-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I build an IDEX printer using 

- BTT Octopus V1.1
- BTT TFT7" V3
- BTT TMC 2209 V1.2 running in SPI mode on all motors
- Nema 17 steppers with 200 full steps per rotation

I have one motor on z axis, and two motors in Y-axis driven by one stepper driver. For that i swapped PINS of y-and z-axis Each x-axis (X1, X2) has a stepper motor.

I based my configuration on the configuration provided at https://enderidex.com
I got all axis working. I got the physical endstops working. ( I was unable to get sensorless homeing to work.) 

The issue I'm describing here: when the X1-axis moves 1 cm and switchs to the X2-axis it will move 2 cm. If I change the steps per mm on x. Both axis change, but the ratio will always be 1:2. I use the move buttons on the TFT (no gcode)

I assumed that the TMC driver for X2 is not configured correctly. I tried to include the TMC settings for X2 and E2 in configuration_adv.h manually, but it did not make any difference.

Connections on the board are as following:

Motor 0: X1
Motor 1: Z (BL 1st)
Motor 2_1: Y_Right (BL 1st)
Motor 2_2: Y-Left (SW 1st)
Motor 3: E1 
Motor 4: E2 
Motor 5: X2 (changed wiring because not possible to invert for IDEX mode in current marlin firmware)

End-Stops:
X1-Min: J27	X_STOP_PIN	PG6
Y-Left-Min: J29		Y_STOP_PIN	PG9
Z-Min:	J31	Z_STOP_PIN	PG10
X2-Max: J33	X_MAX_PIN	PG11

Bed Sens: J44	TEMP_BED_PIN                        PF3
E1 Sens: J45	TEMP_0_PIN                        PF4
E2 Sens: J46	TEMP_1_PIN                          PF5
Bed Heat: Bed_out	HEATER_BED_PIN                      PA1
E1 Heat: HE0	HEATER_0_PIN                        PA2
E2 Heat: HE1	HEATER_1_PIN                        PA3

E1 FAN:	FAN0 (hotend fand, on at certain temperature)
E2 FAN:	FAN1 (hotend fand, on at certain temperature)

LCD7":
EXP1: EXP1
EXP2: EXP2
RS232: 5V/Gnd/TX1/RX1/Reset

### Bug Timeline

I encoutered it the first time (never used that feature before)

### Expected behavior

I expect X1 and X2 axis to travel the exact same distance when giving the same movement command.

### Actual behavior

For X1-axis the real distance traveled is correct, but for X2 it is always the double distance.

### Steps to Reproduce

1. Use the move menu and buttons on the TFT (or gcode)
2. Switch to Extruder 1 (T0)
2. Move 10 mm to the left or right (g1x10f2400 and then g1x20) -> check it actually moved 10 mm in real world 
3. Switch to Extruder 2 (T1)
4. Move 10 mm to the left or right (g1x220 and then g1x230) -> check it actually moved 10 mm in real world

### Version of Marlin Firmware

2.0.9.1

### Printer model

Custom IDEX printer

### Electronics

BTT Octopus V1.1, BTT TFT7" V3, BTT TMC 2209 V1.2

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

[Marlin_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7147678/Marlin_Configuration.zip)
[image]
[image]
[image]

ellensp, 2021-09-11:
you need to enable //#define X2_DRIVER_TYPE TMC2209  not #define E2_DRIVER_TYPE TMC2209	//X2 axis
So the microstepping for x2 is not being set

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-11 by ellensp

MicSchi, 2021-09-11:
Thank you for the fast response. I changed configuration according to your proposal. But it did not solve the issue. It behaves exactly as before by doing the double distance. 

But, I will ask the question in the Marlin Forum again, as this is the right way to go initially.

ellensp, 2021-09-11:
M122 output ?

MicSchi, 2021-09-11:
Just run the M122 command:
[image]
[image]
[image]
[image]

ellensp, 2021-09-11:
where is X2?

MicSchi, 2021-09-11:
Don't know, is there anything I can do about that?

ellensp, 2021-09-11:
Just to confirm you configuration.h now has the following correct?

```CPP
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define X2_DRIVER_TYPE TMC2209
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
//#define I_DRIVER_TYPE  A4988
//#define J_DRIVER_TYPE  A4988
//#define K_DRIVER_TYPE  A4988
#define E0_DRIVER_TYPE TMC2209
#define E1_DRIVER_TYPE TMC2209
//#define E2_DRIVER_TYPE TMC2209	//X2 axis
//#define E3_DRIVER_TYPE TMC2209
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

```

MicSchi, 2021-09-11:
Yes, I changed it accordingly to the above. And to triple check, I again downloaded the latest Marlin source just few minutes ago. And with file compare I transfered my configuration into that source again. Compiled It, copied firmware to SD card. Powercycled the printer and now it is working as expected. I actually don't know what is the difference this time. It can't be in the configuration, must be somewhere else.

But, It is working now. And also get X2 message with M122 now. 

[image]

Thank you again for the rapid help!