MarlinFirmware/Marlin #22753: [BUG] Y axis Home Failure 2.0.9.1 Geeetech A10T, A20, A20M, A20T

andraisr, 2021-09-11:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I recently started testing out 2.0.9.1. I am finding the when homing the Y axis, halfway to the home position there is a slight hesitation, then I get a Homing failure msg on the LCD. This is on several Geeetech Printer models (see title line). I revert back to 2.0.1 and when I home the Y Axis, it is smooth and homes properly. This happens with Octoprint and when directly connected to the computer via USB cable using S3D.

This issue is duplicatable on all the models mentioned. If the bed is close to the home position, it will work. But further than 1/2 way, if fails everytime.

### Bug Timeline

This is a new bug for me. Not sure if this has been reported before.

### Expected behavior

I expect the the Y axis to home when sent a home command.

### Actual behavior

Received a Homing Failure.

### Steps to Reproduce

1. When using the LCD, or Octoprint, or S3D I tell printer to HOME.
2. Movement start.
3. Y axis hesitates, then the printer fails the HOME command. 
4. Receive message from printer and message on display indicating a Homing Failure. 
5. I load 2.0.1 back onto any of the printers, everything works fine.

### Version of Marlin Firmware

2.09.1

### Printer model

A10T, A20, A20M A20T

### Electronics

All stock

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Additional information & file uploads

I repair these printers for GT. Using the stock 2.0.9.1 config files, I get the homing errors. I revert to 2.0.1 and everything works fine. Just attempting to update the fw from the GT standard 1.1.8

thisiskeithb, 2021-09-11:
@Vertabreak frequently works with these machines as well and may have some ideas.

Vertabreak, 2021-09-11:
i have already submitted the fix for this to the configuration repo https://github.com/MarlinFirmware/Configurations/commit/12801003f786254077c45ceb18d4f96314b3e1e5

closed 2021-09-13 by thinkyhead