MarlinFirmware/Marlin #22755: MKS robin e3d v1.1+TMC2130 SPI Fail

Warober, 2021-09-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi I have a problem with MKS robin e3d v1.1.
The problem is that if I load Marlin (2.0.9.1) with the TMC2130, the MKS MINI12864 V3 LCD screen freezes at startup, the Marlin logo appears and crashes. I have tried compiling with TMC2130_STANDALONE and it boots perfectly. I have also tried another screen and the error is the same.
The bug is caused by activating the SPI function of the drivers.
Thanks greetings

### Version of Marlin Firmware

2.0.9.1

### Printer model

Prusa I3 Hephestos

### Electronics

MKS robin e3d v1.1

### Add-ons

MKS MINI12864 V3 

### Your Slicer

Cura

### Host Software

OctoPrint

mks-viva, 2021-09-13:
1. MKS Robin E3/E3D board, the TMC SPI and SD card and EXP2 is same as the pin. So, I suggest you use standard mode. May be marlin can’t handle the situation where they use 1 set of SPI at the same time.

scott-pett, 2022-01-12:
I am also seeing this with BTT Octopus card with BTT TFT35 Display. Enable TMC2130 drivers and I see SD Media init failures.
Revert the drives to A3988 and the thge SD card works fine.
Above just tested on BugFix 2.0.9.3

closed 2026-10-12 by thinkyhead