MarlinFirmware/Marlin #22757: [FR] Detach Servo GCode

Dakkaron, 2021-09-11:
### Is your feature request related to a problem? Please describe.

I am using a servo to switch extruders. But for some reason the servo is constantly twitching minimally (not visible, but very audible humming). This leads to the servo overheating and I already lost one servo due to that.

Using DEACTIVATE_SERVOS_AFTER_MOVE isn't really cutting it for me, since the servo needs to be powered during the tool change. The unpowered servo doesn't have enough resistance for the tool change to work.

### Describe the feature you want

I'd like to have a GCode that I can issue after the tool change to disable the servo when it is not needed any more. This should keep the servo from overheating and using unnecessary power.

ellensp, 2021-09-12:
Twitching servos is normally cheap servos.

Dakkaron, 2021-09-12:
I've tested multiple, also one more expensive one, and all of them twitch (twitch as in vibrate, not as in visible motion) and run hot. And all of them run fine on a RC receiver. I read somewhere (can't find it right now) that the LPC1768, the servo library that Marlin uses and RC servos don't play nicely together, because the signal to the servo jitters too much and that causes the RC servos, which are designed to react fast, to twitch.

Anyways, a solution that even let's the user use the cheapest of servos isn't useless either.

I implemented the GCode to detach the servo, and it completely fixes the problem. Now the servos only hum and vibrate as long as they are needed and after the tool change I send the GCode to detach them, and then they are nice, quiet and cool.

I'll clean it up and create a PR in the next few days.

closed 2021-09-13 by thinkyhead