MarlinFirmware/Marlin #22762: [BUG] Marlin 2.0.9.1 CUSTOM_MENU_MAIN_SCRIPT_DONE "M117 User Script Done" > Not on LCD

roel8032, 2021-09-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

CUSTOM_MENU_MAIN in configuration_adv.h :

#define CUSTOM_MENU_MAIN_SCRIPT_DONE "M117 User Script Done"
The message does not show on the LCD anymore.

Since many variables in the language files have changed, i do believe the problem is from changed variablenames?

Latest bugfix files sept 12 2021.

### Version of Marlin Firmware

2.0.9.1 latest bugfix files sept 12 2021

### Printer model

Infitary M508/ Anet A6.

### Your Slicer

Cura

thinkyhead, 2021-09-13:
Does `M117` work at all on your particular LCD model? Please provide your configurations.

ellensp, 2021-09-13:
Configuration_adv.h has CUSTOM_MENU_MAIN_SCRIPT_DONE  but code expects MAIN_MENU_ITEM_SCRIPT_DONE

Can either update the Config or change the code

In Marlin/src/lcd/menu/menu_main.cpp is
```
    #ifdef MAIN_MENU_ITEM_SCRIPT_DONE
      #define _DONE_SCRIPT "\n" MAIN_MENU_ITEM_SCRIPT_DONE
```

try changing that to

```
  #ifdef CUSTOM_MENU_MAIN_SCRIPT_DONE
      #define _DONE_SCRIPT "\n" CUSTOM_MENU_MAIN_SCRIPT_DONE
```

thinkyhead, 2021-09-13:
Patched.

closed 2021-09-13 by thinkyhead

roel8032, 2021-09-13:
> Does `M117` work at all on your particular LCD model? Please provide your configurations.

Yes, in version 2.0.7. And probably in 2.0.9. download version also.

ellensp, 2021-09-13:
@roel8032 grab the latest bugfix. This has been patched and should work now.

roel8032, 2021-09-13:
> @roel8032 grab the latest bugfix. This has been patched and should work now.
[image]

Confirm it works again.

commit 000d412d (2021-09-13): 🐛 Fix CUSTOM_MENU_MAIN_SCRIPT_DONE

Fix #22762