MarlinFirmware/Marlin #22763: [BUG] POWER_LOSS_RECOVER_ZHOME and Z_SAFE_HOMING is not working correctly

moham96, 2021-09-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
I enabled `POWER_LOSS_RECOVER_ZHOME` while enabling `Z_SAFE_HOMING `which caused this error `POWER_LOSS_RECOVER_ZHOME cannot be used with Z_SAFE_HOMING`.
So I understand that these settings can't be enabled simultaneously because the printer can't ensure that the Z homing will not collide with the model, unfortunately for my setup enabling these settings at the same time is necessary because my X gantry always falls a few millimeters down after the power is lost which causes wrong z height when the print resumes.
So I changed the homing positions to a safe position where it doesn't collide with the model

```
 #define Z_SAFE_HOMING_X_POINT 0
 #define Z_SAFE_HOMING_Y_POINT 10
```

And disabled that error in the `SanityCheck.h` [file](https://github.com/MarlinFirmware/Marlin/blob/3a457c9d13701428a4a276aaf43eb7961e9503a1/Marlin/src/inc/SanityCheck.h#L3398)

The firmware compiles fine now, but the resume function doesn't work properly:

After triggering the power loss function using the `POWER_LOSS_PIN` the printer homes two times(which is weird) and then the print starts from a higher layer than what it was before power loss (i.e if the printer stopped at 0.2 z height it would resume at 0.6 z height)

[Uploading Configurations.zip…]()

### Bug Timeline

not sure

### Expected behavior

I would expect the printer to home **once** and then continue the print exactly where it left of in X, Y, Z axis.

### Actual behavior

The printer is resuming the print job where it left of in the X and Y axis but is the position is not correct in regards to the Z axis

### Steps to Reproduce

1- enable the `POWER_LOSS_RECOVERY` settings as follows:

```
  #define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true
    #define BACKUP_POWER_SUPPLY       
    #define POWER_LOSS_ZRAISE       5 
    #define POWER_LOSS_STATE     HIGH
    #define POWER_LOSS_PULLDOWN
    #define POWER_LOSS_PURGE_LEN   10 
    #define POWER_LOSS_RETRACT_LEN 10
    #define POWER_LOSS_MIN_Z_CHANGE 0.05
    #define POWER_LOSS_RECOVER_ZHOME
    #if ENABLED(POWER_LOSS_RECOVER_ZHOME)
      #define POWER_LOSS_ZHOME_POS { 0, 10 } 
    #endif
  #endif
```
2- enable `Z_SAFE_HOMING` and set the safe position similar to `POWER_LOSS_ZHOME_POS`, also comment out [this line ](https://github.com/MarlinFirmware/Marlin/blob/3a457c9d13701428a4a276aaf43eb7961e9503a1/Marlin/src/inc/SanityCheck.h#L3398) to be able to compile.
3- Start a print job and when it starts printing trigger the power loss by shorting the `POWER_LOSS_PIN` to vcc(take note of the current Z height before triggering the power loss)
4- Restart the printer and click resume

5- You will notice that the printer homes twice and and start printing in the wrong height.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Sep 12 2021 21:53:19)

### Printer model

Creality Ender 3

### Electronics

BigTreeTech SKR 1.4 with TMC 2209s

### Add-ons

BLTOUCH

moham96, 2021-09-16:
It looks like the problem is introduced in this [commit](https://github.com/MarlinFirmware/Marlin/commit/8e56f9366de1b0d78600064aca3f905b4d1d7300#diff-b29d174fc22b838f2cd572ca868d1405b9a14a5f81e48548c9706aa185ff81b8).
The double xyz homing is happening because both of these lines are getting executed
https://github.com/MarlinFirmware/Marlin/blob/8df3e62c89f94f68465ac30e276333940e069ff7/Marlin/src/feature/powerloss.cpp#L413
https://github.com/MarlinFirmware/Marlin/blob/8df3e62c89f94f68465ac30e276333940e069ff7/Marlin/src/feature/powerloss.cpp#L420

thinkyhead, 2021-09-25:
> It looks like the problem is introduced in this commit.

No. This is a feature request, not a bug report. It has never been possible to do safe-homing of Z during resume from power-loss, because that would likely crash into the print, unless the Z safe homing point was set to an edge point. The new feature `POWER_LOSS_RECOVER_ZHOME` adds a position where Z can be homed during resume, but since `G28 Z` always uses the Z safe homing position when `Z_SAFE_HOMING` is enabled, that point is superseded. This feature request is simply requesting that compilation of Power-Loss Recovery to be allowed (via some new option setting) when `Z_SAFE_HOMING` is enabled and set to some point near the edge of the bed, instead of just emitting an #error that forbids it.

thinkyhead, 2021-09-25:
> should i open a separate issue for this ?

Yes! Adding comments here ends up sending email notifications to the person who posted the feature request, so please be courteous and create a new bug report.

resimkulubu, 2022-01-16:
I have bltouch on my printer and when I enable Power Loss Recovery, as soon as it switches to printing, I get this error: reset the printer and start printing.
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/7876605/Marlin-bugfix-2.0.x.zip)

resimkulubu, 2022-01-21:
> Yazıcımda bltouch var ve Güç Kaybı Kurtarma'yı etkinleştirdiğimde, yazdırmaya geçer geçmez şu hatayı alıyorum: yazıcıyı sıfırlayın ve yazdırmaya başlayın. skr 1.4 turbo tmc 2209 bigtreetech tft 3.5- v3
> 
> Güç Kaybı Kurtarma kapatıldığında, sorunsuz bir şekilde düzleşir ve yazdırmaya geçer. lütfen desteğinizden [vazgeçmeyin Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/7876605/Marlin-bugfix-2.0.x.zip)
> 
> Ayarları bu şekilde yaptığımda çalışıyor ama baştan yazdırmaya başlıyor.
> 
> #define POWER_LOSS_RECOVERY #if ENABLED(POWER_LOSS_RECOVERY) #define PLR_ENABLED_DEFAULT false // Güç Kaybını Kurtarma varsayılan olarak etkindir. ('M413 Sn' & M500 ile ayarlayın) //#define BACKUP_POWER_SUPPLY // Güç kaybında stepperleri hareket ettirmek için yedek güç / UPS //#define POWER_LOSS_ZRAISE 2 // (mm) Devam ederken Z ekseni yükselmesi (UPS ile güç kaybında) ) #define POWER_LOSS_PIN -1 // Güç kaybını algılamak için pin. Modülsüz kartlarda varsayılan pini devre dışı bırakmak için -1'e ayarlayın. //#define POWER_LOSS_STATE HIGH // Güç kaybını gösteren pin durumu //#POWER_LOSS_PULLUP tanımlayın // Pullup / pulldown'ı sensörünüze uygun olarak ayarlayın //# POWER_LOSS_PULLDOWN tanımlayın //#define POWER_LOSS_PURGE_LEN 20 // (mm) Devam ederken temizlenecek filamentin uzunluğu //#define POWER_LOSS_RETRACT_LEN 10 // (mm) Başarısız olduğunda geri çekilecek filamentin uzunluğu. Yedek güç gerektirir.

Why is no one supporting?

thisiskeithb, 2022-02-25:
Fixed/implemented in #22828

closed 2022-02-25 by thisiskeithb