MarlinFirmware/Marlin #22764: [BUG] Does not compile without BLTOUCH for CR-10 V3

vdd9474a, 2021-09-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I try to compile a version of Marlin for my CR-10 v3 without BLTOUCH but it does not compile. It works only with BLTOUCH enabled
or there is a lack of documentation/information

### Actual behavior

In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\eeprom.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\watchdog.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1683:4: error: #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
   #error "G29_RETRY_AND_RECOVER requires AUTO_BED_LEVELING_3POINT, LINEAR, or BILINEAR."
    ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o] Error 1

### Steps to Reproduce

Comment "#define CR10V3_BLTOUCH"
And build

### Version of Marlin Firmware

2.0.9.1

### Printer model

Creality CR-10 V3

### Electronics

Stock electronics

### Add-ons

E3Dv6 hotend, PT1000

thinkyhead, 2021-09-13:
You cannot use that option unless you enable `AUTO_BED_LEVELING_3POINT`, `AUTO_BED_LEVELING_LINEAR`, or `AUTO_BED_LEVELING_BILINEAR`. It is not supported for any other bed leveling option.

closed 2021-09-13 by thinkyhead

vdd9474a, 2021-09-13:
I have to enable AUTO_BED_LEVELING_* if I haven't BLTOUCH and want to
disable it in the firmware?
If I haven't BLTOUCH, I don't have auto bed leveling

cp2004, 2021-09-13:
You also have `G29_RETRY_AND_RECOVER` enabled.

vdd9474a, 2021-09-13:
Ok, there is a way to know how to set marlin without bltouch from
configuration file available ? I thought it was just like "comment #define
CR10V3_BLTOUCH in configuration.h"

thisiskeithb, 2021-09-13:
I put in https://github.com/MarlinFirmware/Configurations/pull/567 which disables the BLTouch by default (it's an addon) and also fixed the compile issues when the `CR10V3_BLTOUCH` flag is disabled.

vdd9474a, 2021-09-14:
> I put in MarlinFirmware/Configurations#567

Ok great, what is the version did you use, I have a version issue (it
doesn't compile) event if I git clone the current repository for both,
marlin and configuration files?

vdd9474a, 2021-09-14:
> If a make those modifications on previous version, I can't home on Z

Ok, it works on bugfix 2.0.x
Thanks very much!