MarlinFirmware/Marlin #22765: Combilation exiting with Error

sheminasalam, 2021-09-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have tried both visual code and atom both are exiting marlin 2.0.9.1 with the error 
```
Marlin\src\gcode\motion\G2_G3.cpp:154:7: error: 'MM_PER_ARC_SEGMENT' was not declared in this scope
  154 |       MM_PER_ARC_SEGMENT
      |       ^~~~~~~~~~~~~~~~~~
```

I have tried compiling the previous version and it is successful 

### Version of Marlin Firmware

2.0.9.1

ellensp, 2021-09-13:
You config is not for your version of code.
Your probably attempting to use bugfix or import Configuration.h, which has

if ENABLED(ARC_SUPPORT)
  #define MIN_ARC_SEGMENT_MM      0.1 // (mm) Minimum length of each arc segment
  #define MAX_ARC_SEGMENT_MM      1.0 // (mm) Maximum length of each arc segment

when your code is expecting the older
#if ENABLED(ARC_SUPPORT)
  #define MM_PER_ARC_SEGMENT      1 // (mm) Length (or minimum length) of each arc segment

closed 2021-09-13 by ellensp

neilvangeffen, 2021-09-22:
I have the same issue compiling the latest bugfix with the latest bugfix example for ender 3 max, so doesnt look like a user error to me, looks like code and pre made configs not compatible with each other?

ellensp, 2021-09-22:
@neilvangeffen 
There is a bug with github currently. We cannot fix this, Microsoft has to fix it.

You cannot download bugfix from the download zip button.
It always downloads marlin-2.0.x not marlin-bugfix-2.0.x
Download bugfix using this url and try again https://codeload.github.com/MarlinFirmware/Marlin/zip/refs/heads/bugfix-2.0.x

Or use the git tool and select the bugfix branch with that

Same issue exsists on configuration examples, the download zip link will download import branch. You need bugfix configs from
https://codeload.github.com/MarlinFirmware/Configurations/zip/refs/heads/bugfix-2.0.x

neilvangeffen, 2021-09-22:
i use sourcetree, but maybe i'll try the link and see it it works. To get the latest config, i navigate to the one i want and download the file using the raw button e.g.
"https://github.com/MarlinFirmware/Configurations/raw/bugfix-2.0.x/config/examples/Creality/Ender-3%20Max/Configuration.h"

ellensp, 2021-09-22:
> "https://github.com/MarlinFirmware/Configurations/raw/bugfix-2.0.x/config/examples/Creality/Ender-3%20Max/Configuration_adv.h"
Has 

if ENABLED(ARC_SUPPORT)
#define MIN_ARC_SEGMENT_MM 0.1 // (mm) Minimum length of each arc segment
#define MAX_ARC_SEGMENT_MM 1.0 // (mm) Maximum length of each arc segment

And bugfix code has zero references to MM_PER_ARC_SEGMENT

ellensp, 2021-09-22:
I don't know source tree, but it looks like your using the wrong branch

neilvangeffen, 2021-09-22:
I downloaded your links, copied the config and boot and status screen files, updated platformio and visual studio and keep getting an error C:\users\ned\.platformio\lib\SoftwareSerial\SoftwareSerial.cpp:41:10: fatal error: avr/interrupt.h: No such file or directory

Any chance you can tell me what i'm forgetting this time? I tried googling for a while and tried adding lib_compat_mode=strict but that didnt work either

ellensp, 2021-09-27:
Microsoft has now fixed this bug in github.

neilvangeffen, 2021-09-27:
Thanks for the update! Much appreciated