MarlinFirmware/Marlin #22767: Incorrect ADC resolution

rq3, 2021-09-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

HAL.h for the ST32F1 environment has the ADC resolution set to 10 bits. According to the data sheets for these devices. it should be 12 bits.

### Bug Timeline

probably very old

### Expected behavior

I expected better resoltion from a new hot end sensor.

### Actual behavior

The resoltion of a new hot end sensor was atrocious, so I went looking for the reason.

### Steps to Reproduce

1) Look at the ADC resolution in Hal.h for the ST32F1 environment. It is set to 10 bits.
2) Compare that to the data sheets for the STMicro ST32F1 device family. They are all 12 bit a/d converters.

### Version of Marlin Firmware

all that I have looked at

### Printer model

Anycubic Predator

### Electronics

stock

ellensp, 2021-09-13:
[common_stm32]
platform         = ststm32@~12.1
board_build.core = stm32
build_flags      = ${common.build_flags}
                   -std=gnu++14 -DHAL_STM32
                   -DUSBCON -DUSBD_USE_CDC
                   -DTIM_IRQ_PRIO=13
                   **-DADC_RESOLUTION=12**

This has already been done.

ManuelMcLure, 2021-09-13:
@ellensp `HAL.h` for the STM32F1 unconditionally overrides the setting on the command line so anything that `#includes` `HAL.h` will not see the value set on the command line.

rq3, 2021-09-14:
> 
> 
> @ellensp `HAL.h` for the STM32F1 unconditionally overrides the setting on the command line so anything that `#includes` `HAL.h` will not see the value set on the command line.
> You'd need code in `HAL.h` like
> 
> ```
> #ifndef ADC_RESOLUTION
> #define HAL_ADC_RESOLUTION 10
> #else
> #define HAL_ADC_RESOLUTION ADC_RESOLUTION
> #endif
> ```
> 
> for the `-DADC_RESOLUTION=12` to take effect.

I'm no programmer, but I suspect you could safely invoke 12 bit resolution directly in HAL.h  There are no STM32F1 devices with 10 bit ADC's.

thisiskeithb, 2021-09-14:
> I suspect you could safely invoke 12 bit resolution directly in HAL.h There are no STM32F1 devices with 10 bit ADC's.

This was done on the STM32 HAL as well: https://github.com/MarlinFirmware/Marlin/pull/20562

closed 2021-09-18 by thinkyhead