MarlinFirmware/Marlin #22772: Filament Change "Please wait for prompt" does not go away DWIN_CREALITY_LCD_JYERSUI

doubleO00, 2021-09-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

will not go past Filament Change "Please wait for prompt" 
using DWIN_CREALITY_LCD_JYERSUI

### Expected behavior

To continue after drawing in the filament

### Actual behavior

backs out the old filament, waits for new roll, continue button pulls in new filament, then stops at Filament Change "Please wait for prompt"

### Version of Marlin Firmware

02000902

### Printer model

Creality Ender 3 Max

### Electronics

4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

closed 2021-12-01 by github-actions[bot]

TheWoodser, 2021-12-14:
I am still having this issue.

severandynasty, 2022-02-08:
I too am having this issue, filament load works but you cannot use the purge more filament option as it locks up the LCD.