MarlinFirmware/Marlin #22775: [BUG] Printer Reboots when saving settings on SKR mini E3 v2

GJSchaller, 2021-09-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Starting around the week of Spet. 11th, when I attempt to save a config on my Ender 3 Pro running an SKR mini E3 v2.0, the printer resets, and reboots.  The settings are not saved.

### Bug Timeline

Started around Sept. 11th, 2021, on the 2.0.x branch.

### Expected behavior

Settings to be saved, no reboot.

### Actual behavior

Printer reboots, settings not saved.

### Steps to Reproduce

1) Boot printer
2) Clear EEPROM
3) Flash current firmware
4) Run Z Probe Wizard
5) Save settings - reboot occurs

I believe, but am not sure, this also happens when you issue an M500 via terminal.  Noticed it when running PID, but was not near printer to observe screen.

### Version of Marlin Firmware

2.0.x Bugfix (2.0.9.1)

### Printer model

Creality Ender 3 Pro

### Electronics

BTT SKR mini E3 v2

### Add-ons

TH3D EZABL probe

### Bed Leveling

ABL Bilinear mesh

GJSchaller, 2021-09-15:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7165825/Archive.zip)

Configs attached.

ellensp, 2021-09-15:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22524

closed 2021-09-15 by ellensp