MarlinFirmware/Marlin #22781: [FR] Add comment to HEATER_0_MAXTEMP definition

AbeFM, 2021-09-15:
### Is your feature request related to a problem? Please describe.

Setting "#define HEATER_0_MAXTEMP" to the number you'd like to use will not work, as discussed in https://github.com/MarlinFirmware/Marlin/issues/14993

Setting this number and then not being able to use it can be counterintuitive and is not immediately obvious in the otherwise nicely simplified Configuration.h

### Are you looking for hardware support?

no

### Describe the feature you want

I suggest the existing comment block:
`// Above this temperature the heater will be switched off.
// This can protect components from overheating, but NOT from shorts and failures.
// (Use MINTEMP for thermistor short/failure protection.)`

...be updated to reflect that the maximum temperature the user is able to select is 15 degrees less than this number.

### Additional context

Thanks!

thisiskeithb, 2021-09-15:
You may be using an old copy of Marlin since overshoot settings were extended and  detailed in `Configuration.h`:

https://github.com/MarlinFirmware/Marlin/blob/a1e1555ea1e4a5bb0dabc027b56ffb9fc05739e3/Marlin/Configuration.h#L570-L578

closed 2021-09-15 by thisiskeithb

AbeFM, 2021-09-15:
That is much better, and you're right, I'm on an old version (2.04).

I guess it's for backwards compatibility you don't have overshoot be ABOVE the max set - i.e. temporary, unexpected overshoots go over max temp? At any rate, it's well handled.

Still, a comment specifically in the section where you set the number that tells you not to expect the number you enter here to work seems like it would be a help - and shouldn't require any coding. It appears you're well aware of the situation so I leave it to your judgement. Thanks again.