MarlinFirmware/Marlin #22786: [BUG] SD printing stops when click Printing menu

a0s, 2021-09-16:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Printing from SD hangs if press on "Printing" menu at printer's screen.

### Expected behavior

Printer shouldn't stop printing

### Actual behavior

Printing stops

### Steps to Reproduce

0. Upload big file to SD over Octoprint.
1. Open Octoprint and start printing from SD.
2. Click on Printing menu at printer's screen.

### Version of Marlin Firmware

67d82ff228789408f7f32e6c9af3108c989c0c90 at bugfix-2.0.x

### Printer model

Sapphire Plus

### Electronics

mks nano 1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

thisiskeithb, 2021-09-17:
Which UI are you using? Please attach your config.

a0s, 2021-09-17:
@teemuatlut  I am using TFT_LVGL_UI
My config [config.tar.gz](https://github.com/MarlinFirmware/Marlin/files/7184833/config.tar.gz)

sk8board, 2022-01-01:
I have the same printer, same board, but use TFT_COLOR_UI without OctoPrint. The problem I am having is once the printer starts printing, I can press on menu but the UI does not respond. 

Therefore, the menu to pause the print does show so I cannot pause the print.

I have used the latest bugfix (Dec-30-2021) without any difference to the problem.

closed 2022-06-11 by github-actions[bot]