MarlinFirmware/Marlin #22791: [FR] X-Axis twist compensation

kb-elmo, 2021-09-17:
### Is your feature request related to a problem? Please describe.

On printers similar to the Prusa Mk3 that have two separate rails for the X-axis it's a very common problem that these two axis are not 100% parallel to each other.

This slight "twist" in the X-axis causes the probe Z-offset to be different throughout the length of the X-axis. 
Generating a bed mesh will effectively produce useless garbage since the mesh will sit at an angle that does not match the actual bed geometry of the printer.
The issue gets worse the further the leveling probe (a BLtouch for example) is offset to the nozzle on the Y-axis.

Unfortunately this makes the automatic mesh bed leveling in it's current state pretty much useless for almost every printer that uses this kind of X-axis assembly (in most cases it even makes the problem worse than it is with just manual 4-point leveling).

Here's a drawing (made by MauroGil66 over at https://github.com/andrivet/ADVi3pp/issues/248) that shows the issue pretty good:  
[image]

There have been reports of this in the past already:
https://github.com/MarlinFirmware/Marlin/issues/15206

### Describe the feature you want

My request (or idea) would be to introduce some kind of X-axis twist compensation in the firmware.

Fixing the hardware (i.e. untwisting the X-axis) is pretty much impossible on most of the printers that use this type of gantry assembly since there is simply no good or accurate method of measuring how parallel the axis are and on many printers it's not even possible to move the two rails indipendently from each other.

The (imho) easiest way to implement such a twist compensation would be by setting two different probe Z-offset values.
One for the X-min position and one for the X-max position.
That way the firmware could compensate for the change in probe height when measuring the bed.

Another way would be by setting the X-axis twists as an angle value which would then be used to calculate the Z-offset difference using the provided Y-axis offset for the probe.

Those are just some ideas and since I'm not a very well versed programmer I would leave that to those who know how something like this could be implemented 😃 

### Additional context

Something similar was implemented in the ADVi3pp firmware which is a fork of Marlin

https://github.com/andrivet/ADVi3pp/issues/248

DerAndere1, 2021-09-24:
The author of the original implementation has some financial interest in his development, so in case someone copies it, (s)he should ask the original autor in advance, even though his code is licensed under the terms of GPL 3.0 (mainbord firmware) and partialy under the terms of CC-BY-4.0 (LCD code).

Roxy-3D, 2021-11-20:
This issue has been a big stumbling block for people for a number of years now.   I very much would like to see a good way to adjust for this in the code.    I don't think I would address it with a new mesh.   It probably would be cleaner to correct the twist on the nozzle and Z-Probe directly and and then use that correct position in the generation of any mesh.   

If you have working code, or a limping along version of the code, there are people here that would be interesting in helping to clean it up enough that it can be merged.    Especially if (as stated in Discourse) the only issue is too many global variables were used in its creation.   That won't be too hard to remedy if that is all that is wrong with the prototype code.

andrivet, 2021-12-31:
> The author of the original implementation has some financial interest in his development, so in case someone copies it, (s)he should ask the original autor in advance, even though his code is [...]

Just to clarify: I am the "original author". No need to ask me anything since the code is under GPL. The only things not under GPL (but under CC-BY-4.0) are my design of the screens for the LCD panel. But since Marlin does not support it, there is no problem.

thisiskeithb, 2021-12-31:
Added: https://github.com/MarlinFirmware/Marlin/pull/23238

closed 2021-12-31 by thisiskeithb