MarlinFirmware/Marlin #22792: [BUG] Nozzle crash on power loss recovery restore

mikepthomas, 2021-09-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Using BIGTREETECH SKR 1.4 Turbo control board. Experiencing problem when I resume printing during power loss recovery.
When POWER_LOSS_RECOVERY is enabled, BACKUP_POWER_SUPPLY is enabled and POWER_LOSS_RECOVER_ZHOME is disabled the following abnormal behaviour is observed:

The printer is equipped with bigtreetech ups 12v v2.0. When I switch off the power the printer pauses printing, raises its print head and saves configuration parameters on an SD Card.
When I switch on the power again and select the resume printing option, the printer does homing of XY axes, heats the head bed and heats the head. Then it moves to the XY position where it stopped printing at power off. At the next stage it is about to resume printing. It starts lowering down the nozzle but goes too far crashing into the heat bed.

### Bug Timeline

Old issue as described in https://github.com/MarlinFirmware/Marlin/issues/20516

### Expected behavior

When the printer powers on it is expected to: home XY axes; heat up the head bed and the head; move to the XY position where it stopped printing at power off; lower Z axes to the position when it stopped; resume printing.

### Actual behavior

Instead of returning back to its previous Z position when it stopped printing, the printer keeps lowering the nozzle until it crashes into the heat bed.

### Steps to Reproduce

1. Switch off the power supply. Wait for printer to complete the power off procedure (raise nossle up, switch off)
2. Switch power on.
3. Select resume printing

### Version of Marlin Firmware

https://github.com/mikepthomas/Marlin

### Printer model

Anet A8

### Electronics

Bigtreetech SKR 1.4 Turbo

### Add-ons

Bigtreetech 12v UPS v2.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

The firmware I am using is available on my [GitHub repository](https://github.com/mikepthomas/Marlin)

moham96, 2021-09-19:
You have to monitor the hotend after you switch off the PSU, does the hotend lower a bit due to the gantry weight ? if so then you need to enable the ZHOME option.
I have similar issues with this and found that [this](https://github.com/MarlinFirmware/Marlin/commit/8e56f9366de1b0d78600064aca3f905b4d1d7300) commit causes a lot of issues
could you checkout the commit before it 4a7f5603de04f89aeb47cd68803b810517136acf

mikepthomas, 2021-09-20:
> You have to monitor the hotend after you switch off the PSU, does the hotend lower a bit due to the gantry weight ? if so then you need to enable the ZHOME option.

The gantry does not visibly move down after the PSU switches off or after the Backup Battery depletes the only movement after the PSU switches off is the upwards movement to move the nozzle off the print after power loss.

Unfortunately I have been unable to test the ZHOME option as I have a BLTouch installed with safe Z homing enabled (I do not have a Z min switched installed on the machine and use the probe for z homing) see https://github.com/MarlinFirmware/Marlin/issues/22763

> I have similar issues with this and found that this commit causes a lot of issues
could you checkout the commit before it 4a7f560

I have also tried my config branched off the commit you mention https://github.com/mikepthomas/Marlin/tree/power-loss-debug and I am still having the nozzle drive down through the print into the bed as described in my description above.
I will try this suggestion https://github.com/MarlinFirmware/Marlin/issues/20516#issuecomment-749758948 in the branch later today to see if I have any luck with that, however that code has changed in the latest bugfix so do not think that would work going forward

mikepthomas, 2021-09-21:
> I will try this suggestion #20516 (comment) in the branch later today to see if I have any luck with that, however that code has changed in the latest bugfix so do not think that would work going forward

When trying this, the nozzle restores to completely the wrong place ans then still crashes into the bed...
If I change this line https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L537
to `sprintf_P(cmd, PSTR("G1Z%sF600"), dtostrf(z_raised, 1, 3, str_1));`, the printer starts printing at the level the machine stopped when the power was lost as expected which makes me think `info.zraise` from https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L369 is being calculated incorrectly on my machine. On the power outage function `POWER_LOSS_ZRAISE` is used instead https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L285. I may try to change this when I get some time on the weekend.

mikepthomas, 2021-09-21:
> I will try this suggestion #20516 (comment) in the branch later today to see if I have any luck with that, however that code has changed in the latest bugfix so do not think that would work going forward

When trying this, the nozzle restores to completely the wrong place and then still crashes into the bed...
If I change this line https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L537
to `sprintf_P(cmd, PSTR("G1Z%sF600"), dtostrf(z_raised, 1, 3, str_1));`, the printer starts printing at the level the machine stopped when the power was lost as expected which makes me think `info.zraise` from https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L369 is being calculated incorrectly on my machine. In the power outage function `POWER_LOSS_ZRAISE` is used instead https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp#L285 (I have changed this from the default 2mm to 10mm as decribed in the [UPS user manual](https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/blob/master/miniUPS_V2.0%20(12V)/MINI%20UPS%20V2.0%20user%20manual-2019-11-29.pdf). I may try to change this when I get some time on the weekend.

thinkyhead, 2022-04-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

mikepthomas, 2022-04-22:
Just to update, I never did get to the bottom of the issue and only had 1 failed print due to a power outage since. I unfortunately cannot test the latest bug fix branch as I am currently replacing the SKR 1.4 with an SKR Pico which only works with Klipper as I need the 1.4 for another printer that I am building.

thisiskeithb, 2022-04-22:
Thanks for the follow-up!

If you ever switch back to Marlin and experience this problem again, feel free to open a new issue.

closed 2022-04-22 by thisiskeithb