MarlinFirmware/Marlin #22793: [BUG] Ender3 V2 - Enhanced UI - Screen brightness stuck at half

CRCinAU, 2021-09-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It looks like the LCD Brightness option in the enhanced DWIN display doesn't seem to alter the LCD brightness.

On boot, the LCD seems to go to half brightness, but changing anything in the Advanced Settings does not get reflected in a brightness change of the LCD.

### Bug Timeline

N/A

### Expected behavior

N/A

### Actual behavior

N/A

### Steps to Reproduce

N/A

### Version of Marlin Firmware

bugfix-2.0.x as of 20210917

### Printer model

Ender 3 V2

### Electronics

Stock + BLTouch

### Add-ons

N/A

CRCinAU, 2021-09-18:
@mriscoc - Maybe you have some insight here.

mriscoc, 2021-09-18:
> @mriscoc - Maybe you have some insight here.

That was ok, maybe there was a problem with the last integration: https://github.com/MarlinFirmware/Marlin/pull/22778

doubleO00, 2021-09-18:
As as temp fix I added to \Marlin\src\inc\Conditionals_post.h

#if EITHER(HAS_DWIN_E3V2, IS_DWIN_MARLINUI)
  #define HAS_LCD_BRIGHTNESS 1
  **#define LCD_CONTRAST_MAX  255**
#endif

closed 2021-09-23 by thinkyhead