MarlinFirmware/Marlin #22794: [BUG] Probes with pick-up and drop move sequences need proper support

nionio6915, 2021-09-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am using a detachable Z-Probe and see different movement behaviors for G29 probing, G35 tramming and when probing a single point via G30. The probe pickup and drop off is different depending on which gcode is being executed. 

It is easier to demonstrate the differences in movement patterns with videos. The tramming deploy/stow is more what I was expecting. There are extranious movements and traverses in G29 at the begining and end.  

Probe is currently defined as an Allen key probe. The stepwise movement positions were determined manually (G1 X57 Y30 Z15 ...) and entered into the code block copied from a Delta example config. 

G29 sequence- 
https://photos.app.goo.gl/PDmG4CFRBU4vmH7y6
First move of the deploy is correct: X inline with dock. (0 secs - 15 secs)
skip to 2:20 if you want to skip the majority of the probing
At the end of the stow sequence, it looks like it skips the last position, then high-tails it to the back right corner, then goes back towards the front left corner. . (2:20-2:43)

G35 Tramming sequence
https://photos.app.goo.gl/zLJUegSyyquCxRxW8
There is an extra 'ready' move before the start point of the deploy seqeuence. 

### Bug Timeline

no previous experience

### Expected behavior

I expected the same deploy  and stow sequences regardless of the instructions given to the printer. The exact same motions to deploy and retract.

### Actual behavior

Extra, unexplained or non-instructed motions occur.

### Steps to Reproduce

Run G29 
Run G30 X120 Y120
Run G35 S40 or from the LCD display

### Version of Marlin Firmware

Buffix 2.0.x 2021-08-16

### Printer model

Ender3 Pro

### Electronics

Creality 4.2.2 

### Add-ons

Euclid Probe

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Additional information & file uploads

see above for links to videos.
Current configs: [MarlinEuclidConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/7189296/MarlinEuclidConfigs.zip)
I will try to capture M111 output and post as a followup/edit.

nionio6915, 2021-09-19:
1.  I needed to re-compile with debug support, so a new configs zip file uploaded for completeness.
[EuclidProbeConfigsV2.zip](https://github.com/MarlinFirmware/Marlin/files/7190799/EuclidProbeConfigsV2.zip)

2. Debug output gathered via M111 S39 
   Send: M111 S39
   Recv: echo:DEBUG:ECHO,INFO,ERRORS,LEVELING
[debug09182021.txt](https://github.com/MarlinFirmware/Marlin/files/7190813/debug09182021.txt)

3. I executed the 6 different commands and copy/pasted the output as one file.  I was worried the Octoprint terminal buffer would fill up.  I separated the groups with comment strings to describe what I was doing. 

`
#####
#
# Start of temrinal output from Octoprint
# G28 issued first, then a G35 S40 
#
#####
`
I did:
G28 followed by G29
G30 G(halted/crashed)
G35 S40
Tramming wizard from the LCD, 
Started Tramming wizard from LCD, then G30 X175 Y175 
G30 X175 Y175

4. G30 randomly seem to crash/halt the printer.

nionio6915, 2021-10-02:
the other odd behavior is Marlin seems to return the carraige to the point where a command was issued, instead of the next command- 
If I am at X150 Y150 and issue a G29 command, Marlin will deploy the probe, then return to X150 Y150, then go to the starting point of the mesh pattern

ChinChillaSPb, 2021-10-05:
2.0.9.2 - delta auto cal on kossel linear plus says "stow z-probe" probe v2 after 3rd step in sequence

nionio6915, 2021-12-17:
please reopen this. it is unresolved.

1. trying to use the Z-Probe wizard, the printer will pick up the probe then get confused about which way is +Z and -Z
2. Random 'extra' movements are present at the end of operations like bed probing or bed tramming.
3. Carraige seems to go to the first point of operations before probing sequences, then get the probe. On drop off, it will finish then stow, then return to the end point, then go to the next instruction. 
All this behavior is extranious and superflous. RRF and Klipper are more rational in their operations :(

thisiskeithb, 2021-12-17:
Please comment within the 10-day period next time so the bot doesn’t close the report.

nionio6915, 2022-02-16:
i hope that one of the devs will pick this up. I think its an easy thing to solve!

closed 2022-04-28 by github-actions[bot]

nionio6915, 2022-04-28:
reply reply reply