MarlinFirmware/Marlin #22797: [BUG]MKS robin nano v2, printer halted

Malderin, 2021-09-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Board MKS robin nano v.2, with mksui

After https://github.com/MarlinFirmware/Marlin/pull/22789, robin nano v2 board does not boot. Writes: printer halted please reset.
After canceling https://github.com/MarlinFirmware/Marlin/pull/22789 problem solved

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7189956/Configuration.zip)
.

### Version of Marlin Firmware

Marlin 2.0 bugfix

thisiskeithb, 2021-09-18:
Thanks for including your `platformio.ini`! 

Can you compile with `mks_robin_nano35` to see if these lockups still occur?

Malderin, 2021-09-18:
> Can you compile with `mks_robin_nano35` to see if these lockups still occur?

With default_envs = mks_robin_nano35 and https://github.com/MarlinFirmware/Marlin/pull/22789 no problem.

thisiskeithb, 2021-09-18:
See https://github.com/MarlinFirmware/Marlin/pull/22798

Malderin, 2021-09-18:
> See #22798

Problem solved!

closed 2021-09-19 by thisiskeithb