MarlinFirmware/Marlin #22800: [BUG] MKS E3 V1.1 does not boot since HAL change

jesterhead82, 2021-09-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin versions since the HAL merge ( https://github.com/MarlinFirmware/Marlin/pull/21927 ) are not bootable on my E3 v1.1.

Tried with many versions till on of theis week of Bugfix branch. Falling back using maple allows booting.

Nothing spectacular, as you can see in my attached configs, same config works flawless basically in 2.0xx bugfix till HAL marrge, and works with all releases since Hal merge when using maple.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7190594/Marlin.zip)

### Bug Timeline

Failing since https://github.com/MarlinFirmware/Marlin/pull/21927

### Expected behavior

Boot

### Actual behavior

Stops booting before Marlins Bootscreen

### Version of Marlin Firmware

2.0.xx

### Printer model

MKS E3 v1.1

### Electronics

MKS E3 v1.1

ellensp, 2021-09-19:
Something doesn't add up here....
MOTHERBOARD is set to BOARD_MKS_ROBIN_E3D_V1_1
By default the only env that is valid for that motherboard is env:mks_robin_e3, it doesn't have a maple option.
Sure it can be added, but you made no mention of this.

ellensp, 2021-09-19:
Please config exactly what board you have with a photo of the controller

jesterhead82, 2021-09-19:
Hello,

attached is a pciture of the board.

I am using mks_robin_e3 env till merge of #21927 together with board BOARD_MKS_ROBIN_E3D_V1_1, and mks_robin_e3_maple together with  BOARD_MKS_ROBIN_E3 since merge of #21927.

So mks_robin_e3 and BOARD_MKS_ROBIN_E3D_V1_1 does not boot for me since the merge of #21927 with the attached boards.

[
![20210919_120221](https://use
[image]
r-images.githubusercontent.com/11690785/133939051-c128e8f7-a4d0-4b52-8665-d84a1e163e52.jpg)
](url)

jesterhead82, 2021-09-20:
[image]

Sorry, front picture was missing in upload.

ellensp, 2021-09-21:
Was just looking at circuit diagram. 
Seems like the SPI port should be 2 not 3.

Can you try

```CPP
#
# MKS Robin E3/E3D (STM32F103RCT6) with TMC2209
#
[env:mks_robin_e3]
platform                    = ${common_stm32.platform}
extends                     = common_STM32F103RC_variant
board_build.encrypt         = Robin_e3.bin
board_build.offset          = 0x5000
board_upload.offset_address = 0x08005000
build_flags                 = ${common_STM32F103RC_variant.build_flags}
                              -DTIMER_SERVO=TIM5 -DDEFAULT_SPI=2
build_unflags               = ${common_STM32F103RC_variant.build_unflags}
                              -DUSBCON -DUSBD_USE_CDC
monitor_speed               = 115200
debug_tool                  = stlink

```

mks-viva, 2021-09-22:
@jesterhead82 What you use LCD?
MKS Robin E3D show booting only on LCD12864 or MKSMINI12864

jesterhead82, 2021-09-22:
Hello, 
I`m using a MKS12864**B**, was working till merge of #21927.

mks-viva, 2021-09-22:
Your platformio.ini file env set is what?

jesterhead82, 2021-09-22:
Hello,

i used mks_robin_e3

mks-viva, 2021-09-22:
Maybe you only using an old version of the firmware temporarily

jesterhead82, 2021-09-22:
Hello, 
yes, currently only versions <Jun2021 are working with mks_robin_e3 

Or mks_robin_e3_maple together with BOARD_MKS_ROBIN_E3 if somebody needs to use a newer version

ellensp, 2021-09-22:
Please confirm the terminology.
Lcd not working is not the same as not booting. 
Does it still communicate over usb?  
Does it give any error messages over usb? 

Working on getting this board but eta is months away.

jesterhead82, 2021-09-22:
Hello, 

if I use mks_robin_e3 

- LCD shows "Boot" screen, but does not go ahead
- Connection via USB does not work

With mks_robin_e3_maple
- - LCD shows "Boot" screen, then boots into marlin
- USB connection is working

closed 2021-12-02 by github-actions[bot]

jesterhead82, 2022-01-27:
Tried today Bugfix Branch, seems to boot again with mks_robin_e3.