MarlinFirmware/Marlin #22801: [BUG] BL Touch probing off bed.

EitanSchless, 2021-09-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, when I use any type of bed probing/leveling, the probe goes off the bed then goes down. I've triple checked my probe xyz offsets, and it still doesn't work. For example, I do 'G30 x10 y10', and the probe goes outside the bed. Sorry if I'm not explaining this well. But please help.

### Version of Marlin Firmware

2.0.7.2

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

moham96, 2021-09-19:
Have you enabled `Z_SAFE_HOMING` ?

EitanSchless, 2021-09-19:
No, how can I do that and what will it do?

closed 2021-09-19 by thisiskeithb