MarlinFirmware/Marlin #22807: [BUG] (Config Examples 2.0x Tevo Tornado V1 - Configuration.adv.h Version Issue )

mfleming04, 2021-09-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Information:

- Downloaded latest 2.0x Bugfix Marlin and Configuration Files as of 2021-09-19 - 6:50pm CST
- Built default Marlin and worked fine
- Copy and Replaced the Configuration.h, Configuration_adv.h and _Bootscreen.h from the Config Example folder under Tevo--Tornado-V1 (MKS Base).  No changes or tweaks made to above files at this time.
- Tried to Build
- Received Compile Error:

> Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS E:\=3D_Printer=\=Firmware=\Marlin-2.0.x> platformio run -e mega2560
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.5.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\eeprom.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
Compiling .pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\watchdog.cpp:24:
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\shared\Delay.cpp:24:
Marlin\src\HAL\shared\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\shared\Delay.cpp.o] Error 1
========================================================================================= [FAILED] Took 9.79 seconds ========================================================================================= 

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:09.786
==================================================================================== 1 failed, 0 succeeded in 00:00:09.786 

> 

### Bug Timeline

New

### Expected behavior

Expected to compile by default.

### Actual behavior

Compiling stops, and Errors out.

### Steps to Reproduce

1. Download latest 2.0 Bug fix of Marlin and Configuration files
2. Replace the Configuration.h and Configuration_adv.h from the Tevo Tornado V1 example folder.
3. Try and Built

### Version of Marlin Firmware

2.0.9.1

### Printer model

Tevo Tornado v1

### Electronics

Stock

ellensp, 2021-09-20:
There is a bug with github currently.  
You cannot download bugfix from the download zip button.
It always downloads marlin-2.0.x not marlin-bugfix-2.0.x
Download bugfix using this usrl and try again https://codeload.github.com/MarlinFirmware/Marlin/zip/refs/heads/bugfix-2.0.x
Or use the git tool and select the bugfix branch with that

closed 2021-09-20 by ellensp

mfleming04, 2021-09-20:
Just tried using both links.  It fixed the above issues with the version but now getting.

`Linking .pio\build\mega2560\firmware.elf
The system cannot find the path specified.
*** [.pio\build\mega2560\firmware.elf] Error 1`

ellensp, 2021-09-20:
delete Marlin/.pio directory and try again.

ellensp, 2021-09-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

mfleming04, 2021-09-20:
Nope no go still, but since this is no longer the above which I thought was a bug in the github original files (they should run by default without errors) I have posted to the facebook group like you mentioned.   I may just go back to Arduino for now.  
Thank you.

thisiskeithb, 2021-09-21:
> There is a bug with github currently. We cannot fix this, Microsoft has to fix it.

I filed a ticket and they are aware of the issue and have engineers looking into it.

viveksadhwani, 2021-09-24:
you guys are amazing, thanks