MarlinFirmware/Marlin #22808: [BUG] echo:  M420 S0 Z10.00too much inactive time - current command: OFF

javlonsodikov, 2021-09-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While sending command M503, I see line `echo:  M420 S0 Z10.00too much inactive time - current command: OFF`
how did this happen?

>>> M503 
SENDING:M503
echo:; Linear Units:
  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E400.00
echo:; Max feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E60.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1500.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z10.00too much inactive time - current command: OFF
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; Hotend PID:
echo:  M301 P15.94 I1.17 D54.19
echo:; Bed PID:
echo:echo:  M304 P251.78 I49.57 D319.73
echo:; Filament load/unload:
echo:  M603 L538.00 U555.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S0 ; Sensor
OFF

### Bug Timeline

Latest marlin version

### Expected behavior

echo:  M420 S0 Z10.00

### Actual behavior

echo:  M420 S0 Z10.00too much inactive time - current command: OFF

### Steps to Reproduce

Printer Anycubic i3 Mega

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x (Sep 20 2021 22:15:29) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic i3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Anycubic i3 Mega

### Electronics

stock

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/7197581/configs.zip)
I see memory management got broken

thisiskeithb, 2021-09-24:
Can you see if the changes in https://github.com/MarlinFirmware/Marlin/pull/22829 fix this issue?

javlonsodikov, 2021-09-24:
```>>> m503
SENDING:M503
echo:; Linear Units:
  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E400.00
echo:; Max feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E60.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1500.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z10.00 ; Leveling OFF
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; Hotend PID:
echo:  M301 P15.94 I1.17 D54.19
echo:; Bed PID:
echo:echo:  M304 P251.78 I49.57 D319.73
echo:; Filament load/unload:
echo:  M603 L538.00 U555.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S1 ; Sensor
ON
```

closed 2021-09-25 by thisiskeithb