MarlinFirmware/Marlin #22810: [BUG] Error, Configuration files are for newer version of marlin

niallfo, 2021-09-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am getting an error when compiling for my ender 5 V1 printer 

In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:83:4: error: #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:83:4: error: #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:83:4: error: #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:78:4: error: #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
   #error "Your Configuration.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:55:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:83:4: error: #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for a newer version of Marlin. Upgrade Marlin or downgrade your Configuration_adv.h."
    ^~~~~
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1

### Bug Timeline

yes

### Version of Marlin Firmware

2.0.x.bugfix

### Printer model

ender 5

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

niallfo, 2021-09-20:
changed location of config files

closed 2021-09-20 by niallfo

viveksadhwani, 2021-09-24:
> changed location of config files

did you change location of config files in marlin workspace folder or you changed your main workspace folde?