MarlinFirmware/Marlin #22812: [FR] on power outage, wait for the ongoing gcode line to complete before shutting down

moham96, 2021-09-21:
### Is your feature request related to a problem? Please describe.

Currently, on power outage the printer saves the block number of the gcode line in the file that was printing and shuts down the printer immediately, on resume the print from that block, running the next gcode line after the one that was ongoing, which may lead to gaps in the model if the previous gcode line was not finished before shutting down.

### Describe the feature you want

My idea is that if the printer has backup PSU, the powerloss feature should wait for the currently ongoing gcode command to finish before shutting down, this will avoid having gaps in the model, the feature should be disabled by default and the user can opt out to enable it if they can guarantee that the backup PSU can provide enough power for enough time to complete a whole gcode command.

ellensp, 2021-09-21:
That would require the power reserves to do so. 
The gcode line could be a extrude from one end of the bed to the other, that would require a lot of backup power.

moham96, 2021-09-21:
> That would require the power reserves to do so.
> The gcode line could be a extrude from one end of the bed to the other, that would require a lot of backup power.

Yes and that why i said that it should be disabled by default and only if the user can be sure that their backup PSU has a lot of power they can enable it

resimkulubu, 2022-01-18:
I have bltouch on my printer and when I enable Power Loss Recovery, as soon as it switches to printing, I get this error: reset the printer and start printing.
skr 1.4 turbo
tmc 2209
bigtreetech tft 3.5- v3
Version of Marlin Firmware
Marlin bugfix-2.0.x (Sep 12 2021 21:53:19)

Printer model
Creality Ender 3

Electronics
BigTreeTech SKR 1.4 with TMC 2209

Add-ons
BLTOUCH

Bed Leveling
UBL

Your Slicer
Cura
When Power Loss Recovery is turned off, it smoothly bed-levels and switches to printing. please do not give up your suppor
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/7886647/Marlin-bugfix-2.0.x.zip)

ellensp, 2022-01-18:
@resimkulubu
Please do not tack you unrelated issue to the feature request

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/qNnSzsug
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).