MarlinFirmware/Marlin #22815: #define LASER_SYNCHRONOUS_M106_M107 breaks Fan control in LCD Menu

THE-BOW, 2021-09-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When i activate the option #define LASER_SYNCHRONOUS_M106_M107 i can no more control the fanspeed on my LCD (CR10_STOCKDISPLAY).
When i turn on the printer the fan runs at full speed but the LCD shows 0%
Changing the value does nothing even if i reclick to change the value again.

Controlling the fanspeed in Octoprint, Pronterface or the connected touchscreen works.

### Bug Timeline

Marlin 2.0.9.1

### Expected behavior

I expect to change the fanspeed :-)

### Actual behavior

Nothing happens except the new value is showing in the status screen.

### Version of Marlin Firmware

2.0.9.1

### Printer model

CR10 and Ender-2

### Electronics

SKR V1.3 (CR10) and SKR Mini E3 V2 (Ender-2)

### Add-ons

BL-touch (on both machines) and Laser (on CR10)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

This is the config for my CR10 - I have the same bahaviour on my Ender-2.
Commented out the save homing section in the healthcheck because my bl-touch is on the right ;-)

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/7205796/Configuration_adv.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/7205797/Configuration.h.txt)
[platformio.txt](https://github.com/MarlinFirmware/Marlin/files/7205801/platformio.txt)

thisiskeithb, 2021-09-21:
> When i activate the option #define LASER_SYNCHRONOUS_M106_M107 i can no more control the fanspeed on my LCD (CR10_STOCKDISPLAY).

Which pin is your laser hooked up to? What about your fan?

THE-BOW, 2021-09-21:
I did no changes in the pins configuration.

The partcooling fan is connected to the first default fan pin controlled by M106 P0
The laser is connected to the 2nd default fan pin controlled by M106 P1

THE-BOW, 2021-10-05:
Hi @thisiskeithb
Any more information required?

ellensp, 2021-10-15:
a wild guess. looking at what M106/M107 does with an without LASER_SYNCHRONOUS_M106_M107 vs what the menu does

Can you give this test a try?  

```DIFF
diff --git a/Marlin/src/lcd/menu/menu_item.h b/Marlin/src/lcd/menu/menu_item.h
index 56138a15bd..f65ed4e5d1 100644
--- a/Marlin/src/lcd/menu/menu_item.h
+++ b/Marlin/src/lcd/menu/menu_item.h
@@ -27,6 +27,11 @@
 
 #include "../../inc/MarlinConfigPre.h"
 
+#if ENABLED(LASER_SYNCHRONOUS_M106_M107)
+  #include "../../module/planner.h"
+#endif
+
+
 void lcd_move_z();
 
 ////////////////////////////////////////////
@@ -476,6 +481,7 @@ class MenuItem_bool : public MenuEditItemBase {
 
   inline void on_fan_update() {
     thermalManager.set_fan_speed(MenuItemBase::itemIndex, editable.uint8);
+    TERN_(LASER_SYNCHRONOUS_M106_M107, planner.buffer_sync_block(BLOCK_FLAG_SYNC_FANS));
   }
 
   #if ENABLED(EXTRA_FAN_SPEED)

```

THE-BOW, 2021-10-15:
> Are you doing both laser burns and 3D printing on the same unit? Never mind I see that in the info now.

Hi @descipher 
im using the CR10 as printer and as Laserengraver too. 
I made a mount for the laser on the left of my hotend and a connector to power it on the FAN1 port.
I published it here -> https://www.thingiverse.com/thing:2997540

Using M106 P1... to fire up the Laser and the regular M106 to set the fan speed for the partcooling while printing.
The partcooling FAN also gives me air for the laser to keep the smoke away...

Never had a problem to control both FAN connectors on the CR10_STOCKDISPLAY until the last two Firmwareupdates...

cbagwell, 2022-06-29:
I thought I'd confirm the issue still exists in the bugfix-2.1.x branch and also applying @ellensp 's patch to menu_item.h still fixes the issue.  I could create a PR if it helps to get this fix in but I'm not up to speed with code yet to comment if its an appropriate fix.

I just got a 24v Creality laser that attaches to an Ender's 24v part fan using a Y cable and so I enabled this fan sync option to get ready to test it.  Before testing the laser, I first went looking for any 3d printing regressions the fan sync option caused and setting fan speed in menu is all I've found.

Once I get into full blown laser testing, I hope to try out #22690 branch and I'll report back there.

thisiskeithb, 2022-06-29:
>  I could create a PR if it helps to get this fix in but I'm not up to speed with code yet to comment if its an appropriate fix.

We're still waiting for #22690 to be merged.

descipher, 2022-06-29:
Fixed in #22690 ping @thinkyhead for merge.

thisiskeithb, 2022-06-29:
Since this bug is fixed in https://github.com/MarlinFirmware/Marlin/pull/22690, I'll close this issue.

closed 2022-06-29 by thisiskeithb