MarlinFirmware/Marlin #22831: [BUG] strange coordinate

tome9111991, 2021-09-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello
With latest version.
I have strange coordinate after auto homing.
also i cant set Z offset for probe it jumps in 1mm steps.

I have a ender 3v2 with e3 turbo and stock display.
Tested with `DWIN_CREALITY_LCD` and `DWIN_CREALITY_LCD_JYERSUI`

[image]

### Version of Marlin Firmware

Lstest bugfix

tome9111991, 2021-09-25:
@thinkyhead 
i guess the bug start at https://github.com/MarlinFirmware/Marlin/commit/5b5a8798f8fe202e86c008088f847a3edb51cc71

i have tried  https://github.com/MarlinFirmware/Marlin/tree/cca3250c3ff8e8ed4277f4918f507ab13913d3b5
this works.

with latest commit from now still same issue.

it looks the . (point) is on wrong place
i have a Probe Z Ofsset by -2.74
LCD show me -27.4

tome9111991, 2021-10-21:
With last bugfix.

only have the issue on ``DWIN_CREALITY_LCD``

working is:

```
DWIN_CREALITY_LCD_ENHANCED
DWIN_CREALITY_LCD_JYERSUI
DWIN_MARLINUI_PORTRAIT
```

The-EG, 2021-10-21:
> Btw.
> Durring compiling i got
> #warning "Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs."
> I use dual Z

This is just an informational "warning," letting you know that it's assuming you want to use the next unused E driver for Z2. If that's the case you can just ignore.