MarlinFirmware/Marlin #22832: [BUG] M154 doesn't report current position

moham96, 2021-09-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

when using `M154` to report position in real time it reports incorrect position, instead of returning the current nozzle position, it returns a position in the future i.e probably the position in the last line loaded into the parser.

### Expected behavior

I expect the printer to report the correct current position in real time

### Actual behavior

The printer reports a position in the future

### Steps to Reproduce

1- enable `AUTO_REPORT_POSITION` and compile the firmware.
2- put this gcode in a file on the sdcard:
```
M190 S60
M109 S210.0 T0
G90
G28
G92 E0
G1 Z5.0 F3000
G0 F6000 X10  Y10   Z20 E1
G0 F6000 X10  Y200  Z20 E2
G0 F6000 X60  Y200  Z20 E2
G0 F6000 X60  Y10   Z20 E3
G0 F6000 X120 Y10    Z20 E3
G0 F6000 X120 Y200   Z20 E4
G0 F6000 X180 Y200   Z20 E4
G0 F6000 X180 Y10    Z20 E5
G91
G1 Z7.0 F3000
G90
G1 Y235.0 X235.0 F3000
M121
```
3- Connect to the printer from a host pc and send this gcode `M154 S2` to turn on auto reporting position
4- Start the print and monitor the serial output, you'll notice that as soon as the printer completes homing it will report the incorrect position (reporting `X:235.00 Y:235.00 Z:27.00 E:5.00` while the head is actually in at `X10  Y10   Z20`)

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Sep 24 2021 20:49:57)

### Printer model

Creality Ender 3

### Electronics

BigTreeTech SKR 1.4 with TMC 2209s

### Add-ons

BLTOUCH

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/7227772/Configurations.zip)

thinkyhead, 2021-09-25:
That is fine. It reports the `current_position` variable for the benefit of the display, and it doesn't need to be accurate. If you very much need something closer to the current stepper position, calculated from stepper steps in motion, use `M114` with extended capabilities.

closed 2021-09-25 by thinkyhead

moham96, 2021-09-25:
It might be useful to add the `D` and `R` options to the `M154` gcode to enable the reporter to report realtime positions