MarlinFirmware/Marlin #22836: [BUG] Marlin 2.0.9.1. for Anycubic I3 Mega Compiling error

Digifan, 2021-09-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.9.1 could not be compiled for old Trigorilla board, but it compiled successfully for the Trigorilla v1.1. board with existing examples of configuration files
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7230313/Configuration.zip)
. 

### Bug Timeline

23 september 2021

### Expected behavior

sccessfully build

### Actual behavior

it could not build

### Steps to Reproduce

1. Start VS code with PlatformIO and Auto build Marlid extentions
2. Open Marlin 2.0.9.1 folder and replace configuration files by files as attached
3. try to bui
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7230316/Configuration.zip)
ld

### Version of Marlin Firmware

2.0.9.1

### Printer model

Anycubic I3 Mega

### Electronics

stock 

### Add-ons

no

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Cura

ellensp, 2021-09-26:
Compiles without issue on bugfix with updated Configs
[Bugfix version of Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7230863/Configuration.zip)

closed 2021-09-26 by ellensp

ellensp, 2021-09-26:
This was found and fixed some time ago. https://github.com/MarlinFirmware/Marlin/issues/22254