MarlinFirmware/Marlin #22838: [BUG] 12 Bit Thermistor Tables throw MaxTemp Error

rq3, 2021-09-25:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The latest bugfix successfully shows 12 bit ADC data! However, if a 12 bit thermistor table is created and invoked, and the thermistor table ADC value is set to 12 (from 10) in thermistors.h, the machine instantly throws a "MaxTemp Error, Please Reset" when booted.

A shorted thermistor probe should be ADC=0, and an open thermistor probe should be 4095, in all cases. Turning off all thermal protection does nothing to circumvent this.

### Bug Timeline

very old, I suspect

### Expected behavior

I expect Marlin to report the temperature it is reading.

### Actual behavior

Marlin reports "MaxTemp Error, Please Reset", even when a resistor of a value that should result in a perfectly acceptable temperature is substituted for the thermistor,

### Steps to Reproduce

1) Change the thermistors.h adc value from 10 to 12 
2) Compile, load, and boot

### Version of Marlin Firmware

today's bugfix (09/25/2021, 6:14PM EST)

### Printer model

Anycubic Predator

### Electronics

stock

### Add-ons

none

ManuelMcLure, 2021-09-25:
Thermistor tables should always be 10 bit. 12 bit thermistor tables are not supported.

rq3, 2021-09-26:
> 
> 
> Thermistor tables should always be 10 bit. 12 bit thermistor tables are not supported.

Then that should be fixed. Marlin claims to support 32 bit processors, which generally all have 12 bit ADC capability, primarily used for temperature measurement. When I recently pointed out that Marlin did not use 12 bit ADC capability in the STM series processors, it was rapidly fixed, after going un-noticed for quite some time (years).

Therefore, the thermistor table capability should keep up with the times, and not be brushed off as an "it's always been that way, live with it". 

Frankly, I'm surprised that Marlin even has thermistor tables. Configuration.h should have a short selection of sensor type, R values at 2 or 3 temps, including min and max temps, and a beta factor. It would then develop it's own table from the standard Steinhart-Hart equations as needed.

Marlin basically does just this for platinum RTD sensors. It just doesn't do it very well.

ellensp, 2021-09-26:
@rq3 we still support 8 bit controllers. Which cannot do the math real time thus the use of tables. and are still 10bit ADC's

ManuelMcLure, 2021-09-26:
All 12 bit ADC values are normalized to 10 bits when they are looked up in the tables.

ManuelMcLure, 2021-09-26:
> Frankly, I'm surprised that Marlin even has thermistor tables. Configuration.h should have a short selection of sensor type, R values at 2 or 3 temps, including min and max temps, and a beta factor. It would then develop it's own table from the standard Steinhart-Hart equations as needed.

Are you volunteering to support all of this and generate those values for the dozens of thermistors that are out there, and fix any when users complain that their Marlin is not keeping the proper temperature when the thermistor values that have worked for years suddenly stop working?

rq3, 2021-09-26:
> 
> Thermistor tables should always be 10 bit. 12 bit thermistor tables are not supported.
> All 12 bit ADC values are normalized to 10 bits when they are looked up in the tables.

@ManuelMcLure, call me confused! Which is it? 12 bit tables are not supported, or they are supported but "normalized" to 10 bits when looked up in the tables? It can't be both.

In either case, a 10 bit resolution table (1023 counts) is compressed into an ADC input capable of 12 bit measurement (4095 counts), losing 75% of resolution. The result is evident when plotting the M155 output of a 1047 platinum RTD sensor. The temperature readings bounce, sometimes in excess of 25 degrees. This is not real, but an artifact of the way Marlin uses and processes the table. 

Again, just because something "appears to work" is not sufficient excuse to accept it when it becomes apparent that, in fact, it does NOT work as intended. Simple selection of thermistors #1, #5, and PT1000 #1047, and the use of appropriate precision resistors in place of the sensors themselves will illustrate the issue. Just graph the results in Repetier or the like.

ManuelMcLure, 2021-09-26:
The values read from a 12 bit ADC are normalized to 10 bits before being read in the thermistor table.

rq3, 2021-09-26:
I created a custom 1047 PT1000 table that doesn't invoke the thermistors.h platinum RTD equations. Although it is a 10 bit table, it reduced the temperature plot variation from about 25 degrees to about 0.5 degree, even though it uses only about 200 ADC counts out of the available 1024, due to the flatness of a platinum RTD response.

I'll pursue this further after I install my RTD amplifier, but am closing this for now.

closed 2021-09-26 by rq3